commonsguy/cwac-cam2 #190: CameraActivity is not started The following steps I followed which are as given below:

DanishAmjad12, 2016-04-27:
CameraActivity is not started The following steps I followed which are as given below:

1). First goto settings -> Developer Option -> Checked do not keep activities.
2). Start your application and click on camera button its shows the cameraActivity.
3). Then press home button the application is in OnPause state.
4). Then go back to your application its showing progressbar not cameraActivity
5). Below is the exception detail

```
FATAL EXCEPTION: main Process: com.test.android.test:cwac_cam2, PID: 22841
java.lang.RuntimeException: Unable to stop activity {com.test.android.test/com.commonsware.cwac.cam2.CameraActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.commonsware.cwac.cam2.CameraController.postError(int, java.lang.Exception)' on a null object reference
at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:3670)
at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:3724)
at android.app.ActivityThread.access$1400(ActivityThread.java:151)
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1357)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:135)
at android.app.ActivityThread.main(ActivityThread.java:5254)
at java.lang.reflect.Method.invoke(Native Method)
at java.lang.reflect.Method.invoke(Method.java:372)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.commonsware.cwac.cam2.CameraController.postError(int, java.lang.Exception)' on a null object reference
at com.commonsware.cwac.cam2.CameraFragment.shutdown(CameraFragment.java:267)
at com.commonsware.cwac.cam2.AbstractCameraActivity.onStop(AbstractCameraActivity.java:266)
at android.app.Instrumentation.callActivityOnStop(Instrumentation.java:1277)
at android.app.Activity.performStop(Activity.java:6142)
at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:3665)
at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:3724) 
at android.app.ActivityThread.access$1400(ActivityThread.java:151) 
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1357) 
at android.os.Handler.dispatchMessage(Handler.java:102) 
at android.os.Looper.loop(Looper.java:135) 
at android.app.ActivityThread.main(ActivityThread.java:5254) 
at java.lang.reflect.Method.invoke(Native Method) 
at java.lang.reflect.Method.invoke(Method.java:372) 
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903) 
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698) 
04-27 03:37:27.381 22425-22425/com.test.android.test D/test: destroyed
```

`This same issue is happening in your demo and playground application as well`

```
Brand: Google
DeviceModel: Nexus 5
Product: hammerhead
Manufacturer: LGE
Android Version: 6.0.1
```

DanishAmjad12, 2016-04-27:
This issue is also happening on both cases as follows

- Dont keep activities option enabled in developer option

- When device is running low on memory and android OS is killing the activity

Another device on which this issue has been reproduced
```
Brand:samsung
Device Model:SM-G530H
Product: fortunave3gxx
OS: 4.4.4

```

commonsguy, 2016-05-01:
Addressed in v0.5.9 -- thanks!

closed 2016-05-01 by commonsguy

DanishAmjad12, 2016-05-02:
I faced another issue the following logs are as given below.

```
java.lang.RuntimeException: Unable to stop activity {com.servicelive.android.servicelivepro/com.commonsware.cwac.cam2.CameraActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.commonsware.cwac.cam2.AbstractCameraActivity.canSwitchSources()' on a null object reference
    at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:3670)
    at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:3724)
    at android.app.ActivityThread.handleRelaunchActivity(ActivityThread.java:3924)
    at android.app.ActivityThread.access$900(ActivityThread.java:151)
    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1309)
    at android.os.Handler.dispatchMessage(Handler.java:102)
    at android.os.Looper.loop(Looper.java:135)
    at android.app.ActivityThread.main(ActivityThread.java:5254)
    at java.lang.reflect.Method.invoke(Native Method)
    at java.lang.reflect.Method.invoke(Method.java:372)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.commonsware.cwac.cam2.AbstractCameraActivity.canSwitchSources()' on a null object reference
    at com.commonsware.cwac.cam2.CameraFragment.canSwitchSources(SourceFile:468)
    at com.commonsware.cwac.cam2.CameraFragment.setVideoFABToNormal(SourceFile:464)
    at com.commonsware.cwac.cam2.CameraFragment.stopVideoRecording(SourceFile:443)
    at com.commonsware.cwac.cam2.CameraFragment.stopVideoRecording(SourceFile:439)
    at com.commonsware.cwac.cam2.AbstractCameraActivity.onStop(SourceFile:263)
    at android.app.Instrumentation.callActivityOnStop(Instrumentation.java:1277)
    at android.app.Activity.performStop(Activity.java:6142)
    at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:3665)
```
After fixing the previous issue which you solved in V0.5.9 now we are facing null pointer exception which is causing our application to crash.

commonsguy, 2016-05-02:
Please review [the instructions for contributing a bug report](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports). I need the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish). I also need the steps necessary, using that code and that hardware, to reproduce the problem.

DanishAmjad12, 2016-05-02:
I have created a new issue regarding above NPE. Please review the link #193