commonsguy/cwac-cam2 #192: CameraDevice was already closed

DanishAmjad12, 2016-05-02:
we are getting this intermittent crash issue 

```
java.lang.IllegalStateException: CameraDevice was already closed
    at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:1482)
    at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession(CameraDeviceImpl.java:425)
    at com.commonsware.cwac.cam2.CameraTwoEngine$InitPreviewTransaction.onOpened(SourceFile:430)
    at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:118)
    at android.os.Handler.handleCallback(Handler.java:739)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:145)
    at android.os.HandlerThread.run(HandlerThread.java:61)
```

**Device Details**
```
Brand: Samsung
DeviceModel: SPH-L720T
Product: jflterefreshspr
Manufacturer: Samsung
Android Version: 5.0.1
```

commonsguy, 2016-05-02:
Please review [the instructions for contributing a bug report](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports). I need the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish). I also need the steps necessary, using that code and that hardware, to reproduce the problem.

DanishAmjad12, 2016-05-02:
I skipped your preview screen and make my own preview screen in which user retry multiple times then this issue raises, on Retry functionality I opened again your cameraActivity. & I'll make a sample project for you.

```
  Intent cameraIntent;
        cameraIntent = new CameraActivity.IntentBuilder(this)
                .skipConfirm()
                .focusMode(FocusMode.CONTINUOUS)
                .facing(Facing.BACK)
                .debug()
                .quality(AbstractCameraActivity.Quality.HIGH)
                .to(mCapturedImageUri)
                .debugSavePreviewFrame()
                .build();
        startActivityForResult(cameraIntent, ACTION_GET_PHOTO_FROM_CAMERA_PREVIEW);
```

DanishAmjad12, 2016-05-02:
This issue happens when camera does not open for a while, and loader is shown on the screen. I have attached a image for you to understand
[image]

Our user have reported that this happens sometimes but not always. As you can see camera is unable to open and when user click back and try to reopen the camera reported crash occurs

I have created a sample project for you as well.

[Demo Project Link](https://drive.google.com/file/d/0B1fc9uDBXpvVYk1NYnV0aFl1YlE/view?pref=2&pli=1)

DanishAmjad12, 2016-05-04:
Any update on the above issue?

commonsguy, 2016-05-04:
I recommend that you work with the camera APIs directly. That way, I am not impeding your progress.

Otherwise, I will look at this when I next turn my attention to this library. If we are both lucky, that will be Sunday.

aftabsikander, 2016-05-04:
I have faced this issue as well not always but it is happening rarely.

**Device Detail:**
```
Brand: Google
DeviceModel: Nexus 5
Product: hammerhead
Manufacturer: LGE
Android Version: 6.0.1
```

> I will look at this when I next turn my attention to this library. If we are both lucky, that will be Sunday.

As per my understanding root cause of this issue may be around `CameraDevice.StateCallback` implementation.

commonsguy, 2016-05-08:
Once again, I cannot reproduce the problem. I have made some changes to more cleanly fail when this exception occurs. With luck, I can pull off issue #196 and it helps with the timing issue here.

commonsguy, 2016-05-08:
Change released in v0.5.10.

closed 2016-05-08 by commonsguy