commonsguy/cwac-cam2 #193: NPE when device is running low on memory

DanishAmjad12, 2016-05-02:
I am facing NPE issue which is caused by your library and its making our application crash. This happens when user tries to take pictures multiple times. I have implemented our custom preview screen which we have given user option for retry (to take picture again).  I'll make a sample project for you so that you can easily reproduce this issue on the above mentioned device

I am currently creating a sample project for you in which it can be easily reproduced this NPE. This happens only when device is running low on memory.

As you can see by this log issue is when intent is null `java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.commonsware.cwac.cam2.AbstractCameraActivity.canSwitchSources()' on a null object reference`

**Device Detail**
```
Brand: Samsung
DeviceModel: SPH-L720T
Product: jflterefreshspr
Manufacturer: Samsung
Android Version: 5.0.1
```
Using following intent when user click retry option
 ```
Intent cameraIntent;
        cameraIntent = new CameraActivity.IntentBuilder(this)
                .skipConfirm()
                .focusMode(FocusMode.CONTINUOUS)
                .facing(Facing.BACK)
                .debug()
                .quality(AbstractCameraActivity.Quality.HIGH)
                .to(mCapturedImageUri)
                .debugSavePreviewFrame()
                .build();
        startActivityForResult(cameraIntent, ACTION_GET_PHOTO_FROM_CAMERA_PREVIEW);
```

**Exception Log**

```
java.lang.RuntimeException: Unable to stop activity {com.test.android.test/com.commonsware.cwac.cam2.CameraActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.commonsware.cwac.cam2.AbstractCameraActivity.canSwitchSources()' on a null object reference
    at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:3670)
    at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:3724)
    at android.app.ActivityThread.handleRelaunchActivity(ActivityThread.java:3924)
    at android.app.ActivityThread.access$900(ActivityThread.java:151)
    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1309)
    at android.os.Handler.dispatchMessage(Handler.java:102)
    at android.os.Looper.loop(Looper.java:135)
    at android.app.ActivityThread.main(ActivityThread.java:5254)
    at java.lang.reflect.Method.invoke(Native Method)
    at java.lang.reflect.Method.invoke(Method.java:372)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.commonsware.cwac.cam2.AbstractCameraActivity.canSwitchSources()' on a null object reference
    at com.commonsware.cwac.cam2.CameraFragment.canSwitchSources(SourceFile:468)
    at com.commonsware.cwac.cam2.CameraFragment.setVideoFABToNormal(SourceFile:464)
    at com.commonsware.cwac.cam2.CameraFragment.stopVideoRecording(SourceFile:443)
    at com.commonsware.cwac.cam2.CameraFragment.stopVideoRecording(SourceFile:439)
    at com.commonsware.cwac.cam2.AbstractCameraActivity.onStop(SourceFile:263)
    at android.app.Instrumentation.callActivityOnStop(Instrumentation.java:1277)
    at android.app.Activity.performStop(Activity.java:6142)
    at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:3665)
```

DanishAmjad12, 2016-05-02:
**Link to Demo Project**
[Demo project  link](https://drive.google.com/file/d/0B1fc9uDBXpvVYk1NYnV0aFl1YlE/view?usp=sharing)

Steps to reproduce this issue:

- Rotate device multiple times and take images using retry 

- Run lots of application in stack which causes device to stay in low memory

- After multiple steps of point 1 we observed the above crash

This issue is seen when device is running low on memory.

**Second step is to reproduce this issue easily**

Just make `getIntent() as null` when checking for `canSwitchSources` method in `AbstractCameraActivity`

DanishAmjad12, 2016-05-03:
The issue is in `Camera Fragment  line #468` which is as below
```
 private boolean canSwitchSources() {
   return(((AbstractCameraActivity)getActivity()).canSwitchSources());
 }
```
issue is in getActivity() which is null due to low memory. Android OS is killing your `AbstractCameraActivity` which is causing this NPE. I was able to verify this issue by changing the above code as below

**Modified Code**
```
private boolean canSwitchSources() {
       AbstractCameraActivity activity = null;
       if (((AbstractCameraActivity) getActivity()) != null) {
           activity = ((AbstractCameraActivity) getActivity());
           activity = null;
       }
       return activity.canSwitchSources();
   }
```
After changing your method I got below NPE which is same as reported earlier.
```
05-03 10:58:05.894 5180-5180/com.commonsware.cwac.cam2.playground:cwac_cam2 E/AndroidRuntime: FATAL EXCEPTION: main
Process: com.commonsware.cwac.cam2.playground:cwac_cam2, PID: 5180
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.commonsware.cwac.cam2.playground/com.commonsware.cwac.cam2.CameraActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.commonsware.cwac.cam2.AbstractCameraActivity.canSwitchSources()' on a null object reference
at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2416)
at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476)
at android.app.ActivityThread.-wrap11(ActivityThread.java)
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:148)
at android.app.ActivityThread.main(ActivityThread.java:5417)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.commonsware.cwac.cam2.AbstractCameraActivity.canSwitchSources()' on a null object reference
at com.commonsware.cwac.cam2.CameraFragment.canSwitchSources(CameraFragment.java:462)
at com.commonsware.cwac.cam2.CameraFragment.onHiddenChanged(CameraFragment.java:157)
at com.commonsware.cwac.cam2.CameraFragment.onCreateView(CameraFragment.java:243)                                                                                                 at android.app.Fragment.performCreateView(Fragment.java:2220)
at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:973)
at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1148)
at android.app.BackStackRecord.run(BackStackRecord.java:793)
at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1535)
at android.app.FragmentController.execPendingActions(FragmentController.java:325)
at android.app.Activity.performStart(Activity.java:6267)
at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2379)
at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476) 
at android.app.ActivityThread.-wrap11(ActivityThread.java) 
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344) 
at android.os.Handler.dispatchMessage(Handler.java:102) 
at android.os.Looper.loop(Looper.java:148) 
at android.app.ActivityThread.main(ActivityThread.java:5417) 
at java.lang.reflect.Method.invoke(Native Method) 
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616) 
```
**Solution**
`You must always check AbstractCameraActivity is not equal to null before calling canSwitchSources`
```
private boolean canSwitchSources() {
        if(((AbstractCameraActivity)getActivity())!=null)
        {
            return(((AbstractCameraActivity)getActivity()).canSwitchSources());    
        }
        return false;
    }
```

commonsguy, 2016-05-03:
> You must always check AbstractCameraActivity is not equal to null before calling canSwitchSources

No. I need to figure out why and how you are triggering `canSwitchSources()` when the fragment is detached from its activity.

DanishAmjad12, 2016-05-03:
> Steps to reproduce this issue:

> Rotate device multiple times and take images using retry

> Run lots of application in stack which causes device to stay in low memory

> After multiple steps of point 1 we observed the above crash

> This issue is seen when device is running low on memory.

This issue was reported by our users with above steps & its reproducible on the above mentioned device.

aftabsikander, 2016-05-04:
> No. I need to figure out why and how you are triggering canSwitchSources() when the fragment is detached from its activity.

I am able to reproduce this issue in `Demo-playground Sample` the trick was to rotate device when user clicked capture button while picture was in process of getting saved.

- Enable device rotation on device

- Click capture icon, while picture is getting saved during capturing,rotate device to any orientation. `now activity would be recreated` and above crash would be reproduced.

commonsguy, 2016-05-08:
@aftabsikander 
> during that process if user rotates its device, a new instances of camera fragment is recreated

The fragment is retained. It is not recreated.

@DanishAmjad12 

I cannot reproduce the problem. I have modified `CameraFragment` to attempt to better isolate it from the hosting activity while it is being set up. This particular crash can no longer occur, though I cannot rule out other possible problems from whatever is going on.

commonsguy, 2016-05-08:
Change released in v0.5.10.

closed 2016-05-08 by commonsguy