commonsguy/cwac-cam2 #194: Request: change visibility of these methods to allow for easier custom UI implementations

batrand, 2016-05-04:
As per the discussion in issue #126, my implementation needed these methods to be made `public`:

1. `AbstractCameraActivity.IntentBuilder` and thus also `CameraActivity.IntentBuilder`, `VideoRecorderActivity.IntentBuilder`:  `buildChooserBaseIntent()`
2. `CameraEngine`: `setPreferredFlashModes()`
3. `CameraController`: `supportsZoom()` and `setZoom()`

commonsguy, 2016-05-04:
Would `protected` be sufficient for `buildChooserBaseIntent()`? I don't mind relaxing the visibility, but at the same time, I want the visibility to make sense. I cannot see how `buildChooserBaseIntent()` makes sense outside of the `IntentBuilder` hierarchy itself.

batrand, 2016-05-04:
My implementation needed to return my custom `CameraFragment`, but the `IntentBuilder` of `CameraActivity` only returns `CameraFragment`. That's the whole reason `CameraActivity` is modified in the first place, and the bulk of changes resides in `CameraFragment`. I'm not really sure if `protected` will work, as I'm at home now and cannot access my work computer, and my home Android Studio installation for some strange reasons cannot build at all. I'll verify tomorrow and will update this then!

vxhviet, 2016-05-04:
I kind of like to implement my own layout for the camera but I have real trouble trying to make `CameraFragment` pointing to a different xml. What I've done so far:

Pointing to `MyVideoRecorderActivity` which is extended from `VideoRecorderActivity`
```java
Intent i = new VideoRecorderActivity.IntentBuilder(HomeActivity.this)
                .quality(AbstractCameraActivity.Quality.HIGH)
                .to(new File(saveFolder, DiskUtils.getVideoName()))
                .updateMediaStore()
                .build();

        i.setClass(this, MyVideoRecorderActivity.class);

        startActivityForResult(i, RequestCode.REQUEST_MY_CAMERA);
```   

Then in `MyVideoRecorderActivity`

```java
public class MyVideoRecorderActivity extends VideoRecorderActivity {

    @Override
    protected MyCameraFragment buildFragment() {
        Log.d("MyVideoRecorderActivity", getOutputUri().toString());

        return(MyCameraFragment.newVideoInstance(getOutputUri(),
                getIntent().getBooleanExtra(EXTRA_UPDATE_MEDIA_STORE, false),
                getIntent().getIntExtra(MediaStore.EXTRA_VIDEO_QUALITY, 1),
                getIntent().getIntExtra(MediaStore.EXTRA_SIZE_LIMIT, 0),
                getIntent().getIntExtra(MediaStore.EXTRA_DURATION_LIMIT, 0)));
    }
}
```
Then in `MyCameraFragment`

```java
public class MyCameraFragment extends CameraFragment{
    @Override
    public View onCreateView(LayoutInflater inflater, ViewGroup container, Bundle savedInstanceState) {
    //trying to inflate it with a different xml but it starts getting complicated
    // mostly because of private methods in CameraController and CameraEngine.
    }
}
```

It can be really simple if I just copy over everything and edit only that one line of code but I prefer to keep the source code untouched in case of future upgrade.Yeah, I kind of like to implement my own layout for the camera but I have real trouble trying to make `CameraFragment` pointing to a different xml. What I've done so far:

Pointing to `MyVideoRecorderActivity` which is extended from `VideoRecorderActivity`
```java
Intent i = new VideoRecorderActivity.IntentBuilder(HomeActivity.this)
                .quality(AbstractCameraActivity.Quality.HIGH)
                .to(new File(saveFolder, DiskUtils.getVideoName()))
                .updateMediaStore()
                .build();

        i.setClass(this, MyVideoRecorderActivity.class);

        startActivityForResult(i, RequestCode.REQUEST_MY_CAMERA);
```   

Then in `MyVideoRecorderActivity`

```java
public class MyVideoRecorderActivity extends VideoRecorderActivity {

    @Override
    protected MyCameraFragment buildFragment() {
        Log.d("MyVideoRecorderActivity", getOutputUri().toString());

        return(MyCameraFragment.newVideoInstance(getOutputUri(),
                getIntent().getBooleanExtra(EXTRA_UPDATE_MEDIA_STORE, false),
[... 82 lines cut ...]
    @Override
    protected MyCameraFragment buildFragment() {
        Log.d("MyVideoRecorderActivity", getOutputUri().toString());

        return(MyCameraFragment.newVideoInstance(getOutputUri(),
                getIntent().getBooleanExtra(EXTRA_UPDATE_MEDIA_STORE, false),
                getIntent().getIntExtra(MediaStore.EXTRA_VIDEO_QUALITY, 1),
                getIntent().getIntExtra(MediaStore.EXTRA_SIZE_LIMIT, 0),
                getIntent().getIntExtra(MediaStore.EXTRA_DURATION_LIMIT, 0)));
    }
}
```
Then in `MyCameraFragment`

```java
public class MyCameraFragment extends CameraFragment{
    @Override
    public View onCreateView(LayoutInflater inflater, ViewGroup container, Bundle savedInstanceState) {
    //trying to inflate it with a different xml but it starts getting complicated
    // mostly because of private methods in CameraController and CameraEngine.
    }
}
```

It can be really simple if I just copy over everything and edit only that one line of code but I prefer to keep the source code untouched in case of future upgrade.

batrand, 2016-05-04:
In my own experience `CameraFragment` is very tied to its own layout, so my extension turned out to be basically a reimplementation of much of the logic in the original. I started with `onCreateView` and trace slowly which needed to be reimplemented in my own layout, and worked up slowly until everything runs. Basically what I did was to try to understand how Mark implements his Fragment by reading the code, and then rewrite it to fit my own layout. So, as an example, this is my `onCreateView()`. As you can see, it mirrors the original code very closely, adding and omitting what I do and don't need.

```Java
@Override
    public View onCreateView(LayoutInflater inflater, ViewGroup container, Bundle savedInstanceState) {
        super.onCreateView(inflater, container, savedInstanceState);

        View inflatedView = inflater.inflate(R.layout.fragment_camera, container, false);

        assignViews(inflatedView);
        setupButtons();

        onHiddenChanged(false);
        setButtonsState(false);

        if(getController()!=null && getController().getNumberOfCameras()>0) {
            prepController();
        }

        return inflatedView;
    }
```

Other examples: 
```Java
private void prepController() {
        LinkedList<CameraView> cameraViews = new LinkedList<>();
        CameraView cv = (CameraView) previewStack.getChildAt(0);
        cv.setMirror(false);
        cameraViews.add(cv);

        for(int i = 1; i < getController().getNumberOfCameras(); i++) {
            cv = new CameraView(getActivity());
            cv.setVisibility(View.INVISIBLE);
            cv.setMirror(false);
            previewStack.addView(cv);
            cameraViews.add(cv);
        }

        getController().setCameraViews(cameraViews);
    }

private void takePicture() {
        Uri output=getArguments().getParcelable(ARG_OUTPUT);

        PictureTransaction.Builder b=new PictureTransaction.Builder();

        if (output!=null) {
            b.toUri(getActivity(), output,
                    getArguments().getBoolean(ARG_UPDATE_MEDIA_STORE, false));
        }

        setButtonsState(false);
        getController().takePicture(b.build());
    }

    private SeekBar.OnSeekBarChangeListener zoomListener = new SeekBar.OnSeekBarChangeListener() {
        @Override
        public void onProgressChanged(SeekBar seekBar, int progress, boolean fromUser) {
            if(fromUser) {
                if(getController().setZoom(progress)) {
                    seekBar.setEnabled(false);
                }
            }
        }

        @Override
        public void onStartTrackingTouch(SeekBar seekBar) {}

        @Override
        public void onStopTrackingTouch(SeekBar seekBar) {}
    };

```

Also thanks for the `i.setClass()` line! I did not know that existed! I will try that tomorrow, and if that works, @commonsguy, `IntentBuilder` may not need to be modified at all!

vxhviet, 2016-05-04:
Well, actually if you read the `CameraActivity` and `VideoRecorderActivity` you'll see Mark has already put this down:

/**
   * Class to build an Intent used to start the CameraActivity.
   * Call setComponent() on the Intent if you are using your
   * own subclass of CameraActivity.
   */

but I used to use `setClass()` method so I use that instead (thanks to all those days dealing with external library :)).

batrand, 2016-05-05:
@vxhviet lol so much for me reading code haha.

@commonsguy: 
1. I just checked and since the `CameraFragment` implementation also called `postError()`, that needed to be made `public` as well.
2. I've changed my implementation to use Intent.setClass(), and it works flawlessly! The IntentBuilder edits are no longer needed. 

I've edited the original post for easier referencing.

vxhviet, 2016-05-05:
Glad my problem could help yours #end_sarcasm :D

commonsguy, 2016-05-08:
Change released in v0.5.10.

closed 2016-05-08 by commonsguy