commonsguy/cwac-cam2 #195: Activity open then close instantly on LG G4

MarcoGuerrieri, 2016-05-05:
Hi, 
I've found a bug while developing my application.
In the library of Yazeed44 "MultiImagePicker" that use cwac-cam2 (but also trying to use cwac-cam2 by myself) there is a bug that affect the BACK camera.

Infact if I use the intent:
File photo = new File(Environment.getExternalStoragePublicDirectory(Environment.DIRECTORY_PICTURES), "test"+(new Date().getTime())+".jpg");
Intent i=new CameraActivity.IntentBuilder(getContext())
                            .skipConfirm()
                            .facing(Facing.FRONT)
                            .zoomStyle(ZoomStyle.SEEKBAR)
                            .updateMediaStore()
                            .to(photo)
                            .build();

The cwac-cam activity start correctly and there are no problems taking a picture.

If I try to swipe to the back camera (changing Facing.FRONT to Facing.BACK or also by change the camera directly in the cwac-cam activity), the activity instantly close. The app just print few lines of debug

**LOG**
`LOG W/SplitWindow: update focus... 
LOG E/Editor: hideClipTrayIfNeeded() TextView is focused!! hideClipTray()
LOG I/PhoneWindow: [generateLayout] setColorNavigationBar => color=0x ff000001
LOG D/PhoneWindowEx: Ex2. SystemProperties.get result >> #ff000000
LOG D/PhoneWindowEx: [PWEx][generateLayout] setNavigationBarColor2 : colors=0xff000000
LOG I/PhoneWindow: [setNavigationBarColor2] color=0x ff000000
LOG W/SplitWindow: update focus... 
LOG I/SensorManager: registerListenerImpl() [Sensor: LGE Accelerometer, Rate: 200000, SensorEventListener: android.view.OrientationEventListener$SensorEventListenerImpl@f4a1554] by android.view.OrientationEventListener.enable():93
LOG I/SensorManager: registerListenerImpl() [Sensor: LGE Accelerometer, Rate: 200000, SensorEventListener: android.view.OrientationEventListener$SensorEventListenerImpl@802dcfd] by android.view.OrientationEventListener.enable():93
LOG I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@4856ecb time:81033881
LOG E/ELIPSE: App Version Type: Debug Version
LOG W/SplitWindow: update focus... 
LOG D/Editor: setInputTypeforClipTray(): 0
LOG I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@1cdca7a time:81034521
LOG I/SensorManager: removeAllSensors() [Sensor: LGE Accelerometer] by android.view.OrientationEventListener.disable():108
LOG I/SensorManager: removeAllSensors() [Sensor: LGE Accelerometer] by android.view.OrientationEventListener.disable():108
LOG W/SplitWindow: update focus... 
LOG D/Event: No subscribers registered for event class com.commonsware.cwac.cam2.CameraController$ControllerDestroyedEvent
LOG D/Event: No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent`

**DEVICE:**
**Brand:** LG
**DeviceModel:** LG-H815
**Manufacturer:** LG
**Android Version:** 6.0

MarcoGuerrieri, 2016-05-05:
**IMPORTANT UPDATE:**
After different test, it appears that the problem is in the newer version. Infact, using the default version of MultiImagePicker, that include cwac-cam 0.2.3, the camera work perfectly. 
So I also tried to use that 0.2.3 version of the library in my application for create the CameraActivity Intent, and all work as expected.

commonsguy, 2016-05-06:
While not strictly the same bug, I'll reuse it. The LG G4 was having difficulty with the `demo/` app, failing to offer to capture a portrait, rear-facing camera picture. Resolving by moving this device to the `ClassicCameraEngine`, at least for now.

commonsguy, 2016-05-08:
Change released in v0.5.10.

closed 2016-05-08 by commonsguy