commonsguy/cwac-cam2 #196: Try to Keep Camera Open During Configuration Change

commonsguy, 2016-05-08:
The `CameraFragment` is retained, so in theory we should be able to keep the `CameraController` alive and the `CameraEngine` all open during a configuration change, only closing matters when we are being truly destroyed. This may help with some of the timing errors like issue #192. It would also increase responsiveness on the configuration change. However, there is a possibility that it will screw up camera orientation management.

commonsguy, 2016-05-20:
This isn't completely crazy, but it will need work. Specific changes:

- guts of `CameraController#stop()` move into `CameraController#destroy()` (plus handle the exception)
- `AbstractCameraActivity#init()` needs to only configure and create the controller when creating the fragment
- need to ensure `CameraController#open()` still gets called after above change
- need to work out a way to reconfigure existing plugins (replace the plugin?), as `SizeAndFormatPlugin` needs fresh info based on new `CameraView`
- `CameraController#open()` needs to do that reconfiguration of the plugins and still arrange to update the `SurfaceTexture` with the preview (this may require a new `reopen()` or `attach()` entry point in `CameraEngine`, for an existing open camera switching to a new preview)

Look into this more for 0.7.0.

ramdroid, 2017-03-17:
I was just going to start working on this, until I read about your project contribution guidelines you mentioned in the pull request. So it would be great if this issue still can be included into the next version ;)