commonsguy/cwac-cam2 #197: RedMi Note3 fail on set display orientation failed

jinsen47, 2016-05-08:
Could anyone give me a help?
Log is as below.
```
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event: Could not dispatch event: class com.commonsware.cwac.cam2.CameraEngine$OrientationChangedEvent to subscribing class class com.commonsware.cwac.cam2.CameraController
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event: java.lang.RuntimeException: set display orientation failed
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event:     at android.hardware.Camera.setDisplayOrientation(Native Method)
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event:     at com.commonsware.cwac.cam2.plugin.OrientationPlugin$Classic.configureStillCamera(OrientationPlugin.java:149)
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event:     at com.commonsware.cwac.cam2.ClassicCameraEngine$Session.configureStillCamera(ClassicCameraEngine.java:563)
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event:     at com.commonsware.cwac.cam2.ClassicCameraEngine.handleOrientationChange(ClassicCameraEngine.java:355)
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event:     at com.commonsware.cwac.cam2.CameraController.onEventMainThread(CameraController.java:402)
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event:     at java.lang.reflect.Method.invoke(Native Method)
05-08 23:54:54.004 10625-10625/com.github.jinsen47.camerasample E/Event:     at java.lang.reflect.Method.invoke(Method.java:372)
```

commonsguy, 2016-05-08:
@jinsen47 Please review [the contribution guidelines for how to file a bug report](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports). I need:

- the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem
- the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish)
- the steps necessary, using that code and that hardware, to reproduce the problem

commonsguy, 2016-05-29:
Closing as not reproducible.

closed 2016-05-29 by commonsguy