commonsguy/cwac-cam2 #198: Orientation problem on Blackberry Q5

colinfoko, 2016-05-09:
[cam2_RIM_BlackBerry_20160509-174327.zip](https://github.com/commonsguy/cwac-cam2/files/255595/cam2_RIM_BlackBerry_20160509-174327.zip)

My android applicaiton uses cwac-cam2 without any problems.  However, on the Blackberry Q5 (Small Square screen), if I take a photo holding the device in portrait mode, the photo is actually in landscape.  It should be noted that I never see the photo rotated incorrectly.  In the CameraActivity when I have the screen ready to capture a photo, everything looks "squished".  When I then capture the photo and save it to our servers, it looks perfectly fine when viewed from other devices (it's just that it's in landscape).

Is there anything I can do, to force the cameraActivity to use portrait mode?  I assuming that on this device, it's confused about which orientation it's in.

I also installed the demo app and uploaded the zip if that helps.  Thanks!

colinfoko, 2016-05-09:
Also, I was using version 1 of cwac-cam without any of these problems.  However, in v1, I wasn't using CameraActivity back then, obviously.  I was using the SimpleCameraHost back then.

commonsguy, 2016-05-09:
> if I take a photo holding the device in portrait mode, the photo is actually in landscape

This is true of many devices (most?). They set the EXIF `Orientation` header and that's it. See issue #15.

> I was using version 1 of cwac-cam without any of these problems

There was never a version 1 of any of my camera libraries. CWAC-Camera had EXIF normalization, albeit in Java and only when there was sufficient heap space.

> In the CameraActivity when I have the screen ready to capture a photo, everything looks "squished".

I do not know what this means exactly. Are you referring to the preview having the wrong aspect ratio?

> Is there anything I can do, to force the cameraActivity to use portrait mode?

It is using portrait mode. It's just that the device does not save a portrait image.

colinfoko, 2016-05-09:
I guess the real problem, is that on this device, I have the app in portrait mode and then I call this:

`
com.commonsware.cwac.cam2CameraActivity.IntentBuilder b = 
new com.commonsware.cwac.cam2CameraActivity.IntentBuilder(getActivity());
b.skipConfirm();
b.allowSwitchFlashMode();
b.focusMode(FocusMode.CONTINUOUS);
b.quality(AbstractCameraActivity.Quality.HIGH);
b.facing(Facing.BACK);
i outputUri = Uri.parse(mFeedViewModel.getNewPhotoCapturePath());
b.to(outputUri);

Intent intent = b.build();
startActivityForResult(intent, FeedFragment.REQUEST_CODE_CAMERA);
`

At which point I see the CameraActivity very squished horizontally (as if it's forcing landscape dimensions into a portrait-sized view).  Then when I click the button to capture the photo, it captures it with landscape dimensions.  (The resulting photo is fine, and is rendered fine, except that it's a landscape photo when I expected it to be a portrait photo).

commonsguy, 2016-05-09:
In my next round of work on this library, I will see if I can reproduce the problem on a Q10. I have not been testing on BB10 devices. Frankly, I'm stunned the library holds up as well as it apparently is.

colinfoko, 2016-05-09:
Do you have any advice on where I could go digging around to try to fix this? (we are shipping this week ;-)

commonsguy, 2016-05-09:
Off the cuff, not really. I don't know if the issue is with the square display, how BB10 bridges camera-related code, how BB10 bridges sensor code for detecting portrait/landscape, or something else entirely. I have not had to fuss with this sort of issue in many months.

You may wish to detect BlackBerry hardware (e.g., `Build.MANUFACTURER`) and disable this camera feature on those devices.

colinfoko, 2016-05-10:
Thanks.  Yeah, I think you're right.  For our release we decided to do this for blackberry devices: http://developer.android.com/guide/topics/media/camera.html#intents

Quick change, and it's compatible with your API.  Thanks again for your help

colinfoko, 2016-05-10:
If you need any help when the time comes, just let me know.

commonsguy, 2016-05-10:
With luck, I will look at this on Sunday. Whether I can suss out the source of the problem is another matter.

commonsguy, 2016-05-13:
How have you been testing this? I just downloaded a debug build of the `demo-playground/` app onto my Q10 via the Web browser, and the Q10 refuses to install the app.

commonsguy, 2016-06-08:
Closing as not reproducible.

closed 2016-06-08 by commonsguy