commonsguy/cwac-cam2 #204: Nexus 6P camera layout is not scaled correctly

CodeCombiner, 2016-05-18:
Hi 
Nexus 6P CameraView displays wrong scale. It feels like it is zoomed. Photos are made correctly but preview is zoomed. I think it is related to the camera resolution. If you start Nexus 6P camera app you can see that cameraview height is less than display height. I suppose that CWAC library stretches cameraview height to screen height and camera view width goes beyond screen width.

commonsguy, 2016-05-18:
> I suppose that CWAC library stretches cameraview height to screen height and camera view width goes beyond screen width.

Correct. In other words, this is working as intended.

closed 2016-05-18 by commonsguy

CodeCombiner, 2016-05-18:
[image]
[image]

Just in case here are screenshot examples.
Should CWAC library support "shoot what you see" feature. Might I try to create pull request to support this feature?

commonsguy, 2016-05-18:
> Should CWAC library support "shoot what you see" feature.

When I wrote my original CWAC-Camera library, that is how I implemented it. Developers complained, demanding a full-screen preview. Still other developers have demanded square previews (with the original library and with this CWAC-Cam2 library). Next, somebody will want an oval preview, because InstantFace implemented an oval preview, or whatever.

Frankly, if you care about this, you should not be using this library. This library **only** serves as a replacement for `ACTION_IMAGE_CAPTURE` and `ACTION_VIDEO_CAPTURE`. You do not have control over the preview sizing with third-party camera apps. I offer a modest amount of configurability, but I have neither the time nor the interest to support all possible ways of implementing a camera.

> Might I try to create pull request to support this feature?

You are certainly welcome to fork the library and add the capability to your fork. But implementing what you want is going to take a fair bit of code. For example, IMHO the FABs will have to be replaced with some new UI, as IMHO the FABs will look stupid in a non-full-screen preview, partially floating over black background and partially floating over the preview. That will require a rewrite of `CameraFragment` and new layouts. I cannot accept large pull requests, as is outlined in [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-code).

wwmobile, 2016-11-07:
Hi,
I'd like to avoid the behavior described by @CodeCombiner. Did you realized something to "shoot what you see"?

Regards

commonsguy, 2016-11-07:
> I'd like to avoid the behavior

Use the camera APIs directly.

> Did you realized something to "shoot what you see"?

No.

wwmobile, 2016-11-07:
Thank you for your answer.
Do you think can I combine camera APIs and cwac2 code? 
I wrote my entire application by your code, I found it really useful and clear and it would be a disaster to rewrite the entire application.

Regards

commonsguy, 2016-11-07:
> Do you think can I combine camera APIs and cwac2 code? 

Not really. [As I wrote previously](https://github.com/commonsguy/cwac-cam2/issues/204#issuecomment-220011911), "you are certainly welcome to fork the library and add the capability to your fork".

> it would be a disaster to rewrite the entire application

Then perhaps you should not worry about this behavior. Had you used `ACTION_IMAGE_CAPTURE`, you would have no control over preview sizes and have limited control over picture sizes. The behavior of the user's chosen camera app would vary. Some might work how you prefer; others might not. That would be up to the developers of the world's camera apps.