commonsguy/cwac-cam2 #208: Clients reporting the camera activity immediately closing on LG devices

danielgindi, 2016-05-23:
On multiple devices... They open the camera - it shows up and immediately disappears.

Any known issue?

commonsguy, 2016-05-23:
See issue #195, issue #141, and issue #186. I cannot provide further assistance without the details outlined in [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md).

closed 2016-05-23 by commonsguy

danielgindi, 2016-05-23:
Thank you. There's not much more information that I could capture, but I've seen it in a recorded video.
I'll look into those issues!

danielgindi, 2016-05-25:
More information:
The most problematic device was the LG4, and it was with `cwac-cam2 0.5.11`.

Then I did something stupid. I have compiled `master` branch and released to production. It solved their problems...

commonsguy, 2016-05-25:
@danielgindi Most likely, that is #186. Version 0.6.0 of the library should be out within a week, barring complications, containing that change.