commonsguy/cwac-cam2 #209: Out of memory issues when taking picture

danielgindi, 2016-05-26:
Any way to avoid these? They occur on Samsung Galaxy 5

```
java.lang.OutOfMemoryError: Failed to allocate a 63489036 byte allocation with 16777216 free bytes and 46MB until OOM
	at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
	at android.graphics.Bitmap.nativeCreate(Native Method)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:939)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:912)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:843)
	at com.commonsware.cwac.cam2.ImageContext.rotateViaMatrix(ImageContext.java:242)
	at com.commonsware.cwac.cam2.ImageContext.getJpeg(ImageContext.java:103)
	at com.commonsware.cwac.cam2.JPEGWriter.process(JPEGWriter.java:84)
	at com.commonsware.cwac.cam2.PictureTransaction.process(PictureTransaction.java:38)
	at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:466)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
	at java.lang.Thread.run(Thread.java:818)
```

```
java.lang.OutOfMemoryError: Failed to allocate a 8388396 byte allocation with 4808638 free bytes and 4MB until OOM
	at java.io.ByteArrayOutputStream.expand(ByteArrayOutputStream.java:91)
	at java.io.ByteArrayOutputStream.write(ByteArrayOutputStream.java:201)
	at java.io.BufferedOutputStream.flushInternal(BufferedOutputStream.java:185)
	at java.io.BufferedOutputStream.flush(BufferedOutputStream.java:85)
	at java.io.FilterOutputStream.flush(FilterOutputStream.java:88)
	at com.android.mms.exif.ExifInterface.writeExif(ExifInterface.java:831)
	at com.commonsware.cwac.cam2.ImageContext.getJpeg(ImageContext.java:110)
	at com.commonsware.cwac.cam2.JPEGWriter.process(JPEGWriter.java:84)
	at com.commonsware.cwac.cam2.PictureTransaction.process(PictureTransaction.java:38)
	at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:466)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
	at java.lang.Thread.run(Thread.java:818)
```

kengura, 2016-05-26:
You could skip image normalization with skipOrientationNormalization option in CameraFragment.newPictureInstance() method.

If you really nead to rotate images check this links:
http://stackoverflow.com/questions/12044674/android-rotate-image-without-loading-it-to-memory
http://stackoverflow.com/questions/8294869/rotating-images-on-android-is-there-a-better-way

Another way to solve this issue - increase heap size with android:largeHeap="true" option in manifest file
or decrease taken picture size.

commonsguy, 2016-05-26:
@danielgindi Beyond what @kengura notes, I do need to more gracefully handle these exceptions. I was focused on OOM elsewhere and did not consider this path.

commonsguy, 2016-05-29:
Addressed in v0.6.0.

closed 2016-05-29 by commonsguy