commonsguy/cwac-cam2 #214: Fatal Exception: java.lang.IllegalStateException: Activity has been destroyed

alecio, 2016-06-10:
Got this crash on 'CameraActivity.removeFragments':

Fatal Exception: java.lang.IllegalStateException: Activity has been destroyed
       at android.app.FragmentManagerImpl.enqueueAction(FragmentManager.java:1433)
       at android.app.BackStackRecord.commitInternal(BackStackRecord.java:687)
       at android.app.BackStackRecord.commit(BackStackRecord.java:663)
       at com.commonsware.cwac.cam2.CameraActivity.removeFragments(CameraActivity.java:233)
       at com.commonsware.cwac.cam2.CameraActivity.access$000(CameraActivity.java:31)
       at com.commonsware.cwac.cam2.CameraActivity$2.run(CameraActivity.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:224)
       at android.app.ActivityThread.main(ActivityThread.java:5526)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

commonsguy, 2016-06-10:
As is noted in the project's [contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md), I need:

- the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem
- the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish)
- the steps necessary, using that code and that hardware, to reproduce the problem

commonsguy, 2016-07-16:
Closing as not reproducible.

closed 2016-07-16 by commonsguy