commonsguy/cwac-cam2 #216: Cwac cam2

laltukumar-1473, 2016-06-13:
The camera preview is stretching on orientation change from portrait to landscape using cwac cam2 version 4.0 . How to resolve the issue. Don't want the activity recreation on orientation change.

commonsguy, 2016-06-13:
> The camera preview is stretching on orientation change from portrait to landscape using cwac cam2 version 4.0 .

There is no version 4.0 of this library.

Beyond that, please follow [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md) and provide:

- the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem
- the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish)
- the steps necessary, using that code and that hardware, to reproduce the problem

laltukumar-1473, 2016-06-13:
I have placed  'com.commonsware.cwac:cam2:0.4.+' 
in the gradle file. I am attaching the file for the code.

[CustomFragment.docx](https://github.com/commonsguy/cwac-cam2/files/311661/CustomFragment.docx)
[Manifest file.docx](https://github.com/commonsguy/cwac-cam2/files/311660/Manifest.file.docx)

laltukumar-1473, 2016-06-13:
Device Details: 
Name : HTC 620G
O/S: Android Kitkat

Steps:

1. Open the application camera will open.
2. Now change the orientation from portrait to landscape or vice versa.
3. Look at the camera preview. You will experience the issue.

laltukumar-1473, 2016-06-13:
Please find the entire demo application which I have developed.
[CameraDemoApplication.zip](https://github.com/commonsguy/cwac-cam2/files/311676/CameraDemoApplication.zip)

commonsguy, 2016-06-13:
As is noted in [the documentation](https://github.com/commonsguy/cwac-cam2#basic-usage), I do not offer support for internal APIs ("While there are other public classes and methods in the library, ones that may be exposed as part of a public API in the future, they are not supported at the present time.").

If you can reproduce the problem using the documented and supported API, using the existing activities and fragments without modification, write back with instructions, and I can take a look at it and possibly reopen this issue.

closed 2016-06-13 by commonsguy

laltukumar-1473, 2016-06-13:
Yes, If you try to change the orientation the black page comes then camera preview is oriented.Basically it destroys the view and recreate once again. I don't want to destroy the view.

commonsguy, 2016-06-13:
> Yes, If you try to change the orientation the black page comes then camera preview is oriented

I appear to have introduced a bug somewhere along the line, where dynamic configuration changes are broken on v0.6.0. Follow issue #217 for that.

> I don't want to destroy the view.

You are welcome to write you own code using the camera API to do what you want. While I will work on fixing the bug from issue #217, how I fix that is my decision.