commonsguy/cwac-cam2 #220: Display the 'elapsed time' while recording a video with 'duration limit' set

alecio, 2016-06-15:
Device: LG Nexus 5
Android OS: 6.0.1
Library version: 0.6.0

It seems that the library doesn't support displaying the 'elapsed time' while recording a video. This is quite relevant when the duration limit is set. Is there a way to extend the camera functionality to add this? or should I wait for a new release?

commonsguy, 2016-06-15:
> It seems that the library doesn't support displaying the 'elapsed time' while recording a video.

Are you referring to timers overlaying the preview? Or are you referring to timers embedded in the MP4 of the video itself?

> Is there a way to extend the camera functionality to add this?

Not without modifying the library.

> or should I wait for a new release?

Adding timer overlays as an option would be fairly straightforward. Embedding timers in the video itself would be well outside the scope of this library.

alecio, 2016-06-15:
Thanks for the quick reply.

The timer i'm talking about is not embedded in the MP4, it's just an informative timer overlaying the video preview while the recoding is taking place, just so the user knows when it's reaching a certain 'duration limit'.

It would be nice to have this option, it's indeed not very hard to achieve, but I don't see a way to customize it myself. It would be nice if you can add this option on the coming releases. But let me know if you see alternatives.

Thanks!

commonsguy, 2016-06-19:
Addressed in v0.6.1.

@alecio You can call `chronoType(ChronoType.COUNT_DOWN)` on the `VideoRecorderActivity.IntentBuilder` to enable this countdown timer. Note that it will only be used if you also call `durationLimit()` to set a cap on the duration.

closed 2016-06-19 by commonsguy

alecio, 2016-06-22:
it works great @commonsguy! thank you very much!

mourice, 2016-09-16:
chronoType(ChronoType.COUNT_DOWN) it works great only issue is  timer is starting from 3 sec. Any Idea?

commonsguy, 2016-09-16:
@mourice: `COUNT_DOWN` would start at 3 seconds if the duration limit is 3 seconds. Beyond that, if you have further concerns, please [follow the instructions](https://github.com/commonsguy/cwac-cam2#questions) for asking questions and filing issues.

mourice, 2016-09-16:
Thanks for the reply. Actually I am using count up and I have a limit of
60sec.

commonsguy, 2016-09-16:
@mourice I have set up issue #255 for this.