commonsguy/cwac-cam2 #222: Video preview is upside down when using front camera vertically

shalevmo, 2016-06-21:
When I take a video using the front camera and the phone orientation is vertical, the video output is upside down.

commonsguy, 2016-06-21:
Please follow [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md) and provide:

- the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem
- the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish)
- the steps necessary, using that code and that hardware, to reproduce the problem

shalevmo, 2016-06-21:
I have a Nexus 6P , running Android 6.0.1 (MTC19V).
All I did was to take a video using the sample code:

`  Intent i=new VideoRecorderActivity.IntentBuilder(MainActivity.this)
      .quality(Quality.LOW)
      .facing(VideoRecorderActivity.Facing.FRONT)
      .to(new File(testRoot, "test.mp4"))
      .debug()
      .updateMediaStore()
      .build();`

then, when I watch the video in any video player, it is upside down.

commonsguy, 2016-06-21:
There are two portrait orientations. Have you tried both?

shalevmo, 2016-06-21:
only the normal portrait orientation is working for me.
when i try to flip my phone it goes to one of the landscape orientations

commonsguy, 2016-06-21:
I can reproduce the problem on a 6P and will look into it in my next development pulse on this. Thanks!

shalevmo, 2016-06-21:
thank you very much!,  i'll be waiting.
BTW, are there any planes to add flashlight toggle?

commonsguy, 2016-06-21:
I have no idea what a "flashlight toggle" is. If you mean user-selectable flash modes, see issue #126.

commonsguy, 2016-06-27:
Addressed in v0.6.2.

closed 2016-06-27 by commonsguy