commonsguy/cwac-cam2 #223: How to improve the shutter lag?

vxhviet, 2016-06-22:
Hi @commonsguy, I would like to ask is it possible to improve the shutter lag in the classic camera engine (I'm not using the cam2 engine)? Currently it takes about 500ms after the shutter is clicked to actually start capturing. I know hardware is not the limitation as the stock Android Camera app can capture almost instantly.

Device: Nexus 5 running Android 6.0.1, cwac-cam2 0.5.10.

commonsguy, 2016-06-22:
> Currently it takes about 500ms after the shutter is clicked to actually start capturing

How are you measuring this?

vxhviet, 2016-06-22:
Unfortunately not, just an observation on my implementation and your demo. I've also investigated similar camera apps, VSCO has the same lag, Instagram's lag is a bit less but it's still there. So I'm really surprise how the stock Android Camera has that almost instant capturing.

This is important to my app as it targets fast action scene, I'm trying to find a way to at least reduce it to a less noticeable level.

commonsguy, 2016-06-22:
I cannot afford to spend time on an issue for which I will never know if it is resolved.

If you want to grab the source code and give me a more scientific definition of your issue (e.g., "it is taking X milliseconds on average to go from this line of code to that line of code, and that would appear to be where the lag is that I am perceiving"), I can look into it. Provide that information as a comment here, and I can reopen the issue.

closed 2016-06-22 by commonsguy

vxhviet, 2016-06-27:
Hi, thank you for your response. After a quick research, looks like the `Camera2` engine has already has this support via [`TEMPLATE_ZERO_SHUTTER_LAG`](https://developer.android.com/reference/android/hardware/camera2/CameraDevice.html#TEMPLATE_ZERO_SHUTTER_LAG). Too bad I couldn't find anything regarding the `Camera` engine.