commonsguy/cwac-cam2 #229: Video recording session is interrupted if the device is rotated

alecio, 2016-06-29:
Device: Nexus 5
OS: 6.0.1
Library version: 0.6.1

While recording a video (after pressing the record button) if the device gets rotated the recording session is stopped.

We could provide an extra to prevent it from happening, we could fix the screen orientation on the AndroidManifest.xml as a workaround, but ideally it would lock the screen as it was when the user started (eg. keep it Landscape or keep it Portrait according to the previous screen's orientation).

commonsguy, 2016-07-03:
I agree that the orientation should be locked during recording and photo-taking. That's a significant enough behavior change that I wanted to experiment with it first.

v0.6.3 adds `orientationLockMode(OrientationLockMode)` to the `IntentBuilder`. This sets up the overall orientation lock policy: `DEFAULT`, `PORTRAIT`, or `LANDSCAPE`. The latter two lock the orientation for the whole activity lifetime. `DEFAULT` has the existing behavior, with an eye towards it locking the current orientation during photo/video capture in the future.

If you get a chance, play with v0.6.3 a bit. The `demo-playground/` app has preferences for setting the orientation lock mode. So far, this seems to be working OK for me, but I want to let people play around with this for a while before I add in the dynamic `DEFAULT` orientation locking.

alecio, 2016-07-12:
thanks @commonsguy! We're already using the 0.6.3 on our build and it is working well and more importantly it is preventing the video interruption if a rotation takes place in a recording session.

Still the the problem with this approach is that we have to decide the orientation for user prior to the camera view, and usually it isn't a nice user experience.

It would be very much appreciated to see the feature of locking it when the record button is pressed, this way, the user can still rotate and choose the orientation when in idle on the CameraView.

Thanks anyway.!

commonsguy, 2016-07-12:
> Still the the problem with this approach is that we have to decide the orientation for user prior to the camera view, and usually it isn't a nice user experience.

Again, `orientationLockMode()` wasn't designed to address your problem. It was designed to confirm that the technical solution worked. Dynamic locking is the next step.

vxhviet, 2016-08-17:
If you want, you can implement this feature yourself. I've been doing this on our product code and there's  no crash relating to this feature at all so I can say this solution is quite stable.

(This is for version 0.5.11, so you may want to re-test it on 0.6.xx):

On your `VideoRecorderActivity` enable auto orientation with this:

```java
private static void setAutoOrientationEnabled(Context context, boolean enabled) {
        Settings.System.putInt(context.getContentResolver(), Settings.System.ACCELEROMETER_ROTATION, enabled ? 1 : 0);
    }
```

On your `CameraFragment`, you lock and unlock the orientation according to record button with these:

```java
private void lockOrientation(){
        if(getActivity().getResources().getConfiguration().orientation == ActivityInfo.SCREEN_ORIENTATION_LANDSCAPE){
            getActivity().setRequestedOrientation(ActivityInfo.SCREEN_ORIENTATION_LANDSCAPE);
        }

        if(getActivity().getResources().getConfiguration().orientation == ActivityInfo.SCREEN_ORIENTATION_PORTRAIT){
            getActivity().setRequestedOrientation(ActivityInfo.SCREEN_ORIENTATION_PORTRAIT);
        }
    }

    private void unlockOrienation(){
        getActivity().setRequestedOrientation(ActivityInfo.SCREEN_ORIENTATION_SENSOR);
    }
```

coryjthompson, 2016-08-19:
Thanks @vxhviet. 

I used your code in. 
https://github.com/commonsguy/cwac-cam2/compare/master...coryjthompson:issue229

It appears to be working quite nicely.

commonsguy, 2016-11-20:
Addressed in v0.7.0

closed 2016-11-20 by commonsguy