commonsguy/cwac-cam2 #233: Check Additional Runtime Permissions in Utils.validateEnvironment()

commonsguy, 2016-07-02:
It is the app's responsibility to confirm API Level 23 runtime permissions are in place before starting the camera activity. However, not everybody will do this, so we should check for this and fail quickly if the permissions are missing.

In v0.6.3, `Utils.validateEnvironment()` will confirm that we have the `CAMERA` permission, since that is always required.

However, we should also check for:

- `RECORD_AUDIO`, if we are recording video, but this is not needed for still photos
- `WRITE_EXTERNAL_STORAGE`, if that appears to be where we are writing the output

`validateEnvironment()` will need additional parameters for this, so this will be a breaking API change.

commonsguy, 2016-10-15:
The `WRITE_EXTERNAL_STORAGE` part will be difficult to implement reliably, so this commit just handles `RECORD_AUDIO`.

idish, 2016-11-11:
Why would you check in `Utils.validateEnvironment()` that we have the needed permissions if you request it anyways later on in here: 
`if (useRuntimePermissions()) {
      String[] perms=netPermissions(getNeededPermissions());

      if (perms.length==0) {
        init();
      }
      else {
        requestPermissions(perms, REQUEST_PERMS);
      }
    }`? If you check that all the permissions are already satisfied before this executes, there's no real need for this code, right?

commonsguy, 2016-11-11:
> if you check that all the permissions are already satisfied before this executes, there's no real need for this code, right?

I do need to reconcile this. You are correct that this is redundant.

> you mean that the people who use the library will check it beforehand?

Clients should request the permissions before starting the camera activity.

> I think it should be the lib's responsibility.

Only as a fallback, at most, IMHO.

Partially, that is because app developers need to think through when and how they are requesting permissions and should not have third-party code randomly request permissions behind their backs. For example, I have no good means of providing rationale messages for the permissions.

Partially, that is because I cannot guarantee that it will be practical for a library to request permissions in the future, depending on possible changes in Android's permission system in the future.

On the whole, I would prefer to keep the scope of this library down.

idish, 2016-11-11:
> Only as a fallback, at most, IMHO.
>
> Partially, that is because app developers need to think through when and how they are requesting permissions and should not have third-party code randomly request [...]

Agreed! That makes sense.

commonsguy, 2016-11-16:
@idish OK, here is what I have done:

- By default, the library sees if we have the necessary runtime permissions and fails fast if we lack them.

- If you call `requestPermissions()` on the `IntentBuilder`, the library will handle the runtime permission request for you. This is not recommended; apps should request the permissions themselves, to be able to better handle rationale and the user declining to grant the permissions.

idish, 2016-11-16:
@commonsguy Sounds excellent.

commonsguy, 2016-11-20:
Addressed in v0.7.0

closed 2016-11-20 by commonsguy