commonsguy/cwac-cam2 #237: NoClassDefFoundError for EventBus on 4.x devices

vxhviet, 2016-07-06:
Hi, this is more like a bug report for someone running into similar problem.

So I've been using Fabric to investigate bug and for the past couple of weeks it seem like every Android 4.x device encounter this particular crash (I'm using cwac 0.5.11):

```
Fatal Exception: java.lang.NoClassDefFoundError: android/os/PersistableBundle
       at java.lang.Class.getDeclaredMethods(Class.java)
       at java.lang.Class.getDeclaredMethods(Class.java:605)
       at de.greenrobot.event.SubscriberMethodFinder.findSubscriberMethods(SubscriberMethodFinder.java:75)
       at de.greenrobot.event.EventBus.register(EventBus.java:163)
       at de.greenrobot.event.EventBus.register(EventBus.java:133)
       at com.commonsware.cwac.cam2.AbstractCameraActivity.onStart(AbstractCameraActivity.java:256)
       at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1188)
       at android.app.Activity.performStart(Activity.java:5382)
       at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2279)
       at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2362)
       at android.app.ActivityThread.access$700(ActivityThread.java:168)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1329)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:176)
       at android.app.ActivityThread.main(ActivityThread.java:5493)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:525)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1225)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1041)
       at dalvik.system.NativeStart.main(NativeStart.java)
Caused by java.lang.ClassNotFoundException: Didn't find class "android.os.PersistableBundle" on path: DexPathList[[zip file "/data/app/co.example.myexample-2.apk"],nativeLibraryDirectories=[/data/app-lib/co.example.myexample-2, /vendor/lib, /system/lib]]
       at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:64)
       at java.lang.ClassLoader.loadClass(ClassLoader.java:501)
       at java.lang.ClassLoader.loadClass(ClassLoader.java:461)
       at java.lang.Class.getDeclaredMethods(Class.java)
       at java.lang.Class.getDeclaredMethods(Class.java:605)
       at de.greenrobot.event.SubscriberMethodFinder.findSubscriberMethods(SubscriberMethodFinder.java:75)
       at de.greenrobot.event.EventBus.register(EventBus.java:163)
       at de.greenrobot.event.EventBus.register(EventBus.java:133)
       at com.commonsware.cwac.cam2.AbstractCameraActivity.onStart(AbstractCameraActivity.java:256)
       at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1188)
       at android.app.Activity.performStart(Activity.java:5382)
       at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2279)
       at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2362)
       at android.app.ActivityThread.access$700(ActivityThread.java:168)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1329)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:176)
       at android.app.ActivityThread.main(ActivityThread.java:5493)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:525)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1225)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1041)
       at dalvik.system.NativeStart.main(NativeStart.java)
```
Those devices are:
- Samsung Galaxy Grand 2 (SM-G710) running Android 4.3
- Leotech C55 Negro (LESPH5504B) running Android 4.4.2
- HUAWEI Ascend (Y330-U05) running Android 4.2.2
- Samsung Galaxy Fame (GT-S6810) running Android 4.1.2
- AG CHROME ULTRA running Android 4.4.2
- ARBUTUS AR5 running Android 4.4.2
- HUAWEI Honor 3C (H30-U10) running Android 4.2.2
- OPPO N1 mini (N5111) running Android 4.3
- Zopo ZP980+ running Android 4.2.2
- Samsung Galaxy Tab 4 (SM-T231) running Android 4.4.2
- Samsung Galaxy Tab 4 (SM-T331) running Android 4.4.2
- Samsung Galaxy S4 mini (GT-I9190) running Android 4.4.2
- Samsung Galaxy S III (SPH-L710T) running Android 4.4.2
- Motorola Moto G 4G (XT1039) running Android 4.4.4
- Motorola Droid Ultra (XT1080) running Android 4.4.4
- Pantech Vega (IM-A880S) running Android 4.4.2
- Samsung Galaxy Grand 2 (SM-G7105) running Android 4.4.2
- Acer Iconia One 7 (B1-730) running Android 4.2.2
- Samsung Galaxy Grand Neo (GT-I9060I) running Android 4.4.4
- Samsung Galaxy Tab E (SM-T561Y) running Android 4.4.4
- HUAWEI Ascend (G510-0251) running Android 4.1.1
- MyPhone Agua Rio Lite running 4.2.2
- Motorola Moto G (XT1032) running 4.4.4
- Rivo Phantom PZ8 running  4.4.2

I do follow the instruction to configure proguard but I think I also need to [configure proguard for `EventBus` ](http://greenrobot.org/eventbus/documentation/proguard/) like this:

```
-keepattributes *Annotation*
-keepclassmembers class ** {
    @org.greenrobot.eventbus.Subscribe <methods>;
}
-keep enum org.greenrobot.eventbus.ThreadMode { *; }

# Only required if you use AsyncExecutor
-keepclassmembers class * extends org.greenrobot.eventbus.util.ThrowableFailureEvent {
    <init>(java.lang.Throwable);
}
```

I don't think the last one is necessary as I don't see `AsyncExecutor` being used anywhere in the library.

Let me investigate this in a couple of days to see if this solve the issue.

commonsguy, 2016-07-06:
@vxhviet I suspect that this will not help.

Did you modify `AbstractCameraActivity` in your own copy of the library? Based on [the greenrobot EventBus FAQ](http://greenrobot.org/eventbus/documentation/faq/) (see second entry in "Troubleshooting"), this problem should only occur if you override a method referencing `PersistableBundle`, which I have not done. I test on a wide range of 4.x devices and have not run into this crash.

vxhviet, 2016-07-07:
Well I do add a small addition to the `AbstractCameraActivity` to keep the screen on during recording, like this:

```
protected void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);

    //Todo: [BOOKMARK] keep screen on during recording
    getWindow().addFlags(WindowManager.LayoutParams.FLAG_KEEP_SCREEN_ON);

    Utils.validateEnvironment(this);
```

But I think I've found the problem. On my extension of your `VideoRecorderActivity`, Android Studio must have accidentally generated this stuff:

```
    @Override
    public void onCreate(Bundle savedInstanceState, PersistableBundle persistentState) {
        super.onCreate(savedInstanceState, persistentState);
    }
```

I've removed that and the proguard rules. Let me test for a couple of days and I will report back.

Thank you so much for your assistant.

commonsguy, 2016-07-16:
Closing as being not strictly related to this library. If you determine otherwise, comment and I can reopen the issue if needed.

closed 2016-07-16 by commonsguy