commonsguy/cwac-cam2 #239: E/MediaRecorder: start failed: -38

Ponsen, 2016-07-08:
Hi,

I'n having an issue where I want to record a video coming from an activity which uses the camara, too.

In detail I have a WebRTC (Voip) actitivy where you can take pictures or record videos during a call. When I'm in a call I want to record a video, but this results in the following error:
 
`07-08 11:19:04.044 9841-9841/com.company.product E/MediaRecorder: start failed: -38
07-08 11:19:04.044 9841-9841/com.company.product E/CameraFragment: Exception recording video
                                                                       java.lang.IllegalStateException
                                                                           at android.media.MediaRecorder._start(Native Method)
                                                                           at android.media.MediaRecorder.start(MediaRecorder.java:946)
                                                                           at com.commonsware.cwac.cam2.ClassicCameraEngine.recordVideo(ClassicCameraEngine.java:289)
                                                                           at com.commonsware.cwac.cam2.CameraController.recordVideo(CameraController.java:215)
                                                                           at com.commonsware.cwac.cam2.CameraFragment.recordVideo(CameraFragment.java:457)
                                                                           at com.commonsware.cwac.cam2.CameraFragment.performCameraAction(CameraFragment.java:418)
                                                                           at com.commonsware.cwac.cam2.CameraFragment$1.onClick(CameraFragment.java:243)
                                                                           at android.view.View.performClick(View.java:5698)
                                                                           at android.view.View$PerformClick.run(View.java:22565)
                                                                           at android.os.Handler.handleCallback(Handler.java:739)
                                                                           at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                           at android.os.Looper.loop(Looper.java:158)
                                                                           at android.app.ActivityThread.main(ActivityThread.java:7231)
                                                                           at java.lang.reflect.Method.invoke(Native Method)
                                                                           at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                           at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)`

Recording a video and not beeing in a call works just fine.
Taking a Picture during a call works fine, too.

I've tried adding a delay (3 seks) in AbstractCameraActivity when calling init(), to see if it was a race condition where the camera was not released in time, but I can see the preview and as I said: taking pictures works.

Any Ideas?

Ponsen, 2016-07-08:
Just found out, that the mic is still block by my activity. Releasing the microfone and the camera fixes the issue.

closed 2016-07-08 by Ponsen

Sharabaddin, 2017-10-17:
Fix start record  another service. https://stackoverflow.com/questions/23971817/mediarecorder-start-failed-38. 

Please share your fix. How unblock mic in your activity?

commonsguy, 2017-10-17:
@Sharabaddin This project has been discontinued.