commonsguy/cwac-cam2 #241: Image preview/Image rotated 90 degrees when using PORTRAIT FFF

shahzadyounas, 2016-07-19:
[image]

I am on the latest version 0.6.4 - however still seeing this issue on Galaxy S4 Mini and Lenovo TAB S8-50F . My app requires use of the front facing camera only in portrait mode:

```
 Intent i=new CameraActivity.IntentBuilder(RegistrationStage_5.this)
                        .facing(Facing.FRONT)
                        .facingExactMatch()
                        .to(new File(getExternalCacheDir(), "portrait-front.jpg"))
                        .orientationLockMode(OrientationLockMode.PORTRAIT)
                        .zoomStyle(ZoomStyle.NONE)
                        .updateMediaStore()
                        .build();

                startActivityForResult(i,  REQUEST_PORTRAIT_FFC);
```
However in emulator, and samsung S4 mini and Lenovo Tablet all previews are rotated 90degrees?

Samsung BUILD.PRODUCT: serranoltexx (Samsung S4 Mini CyangenMod)
Lenovo BUILD.PRODUCT: S8-50F

Also Emulator 23 shows same issue.

Have tried with orientation lock removed but makes no difference.

shahzadyounas, 2016-07-19:
Also - Samsung model appears to side shift and crop the image (vs what was shown in the live camera view pre snap).

commonsguy, 2016-07-19:
I am not going to deal with the Samsung, as I only worry about stock ROMs. Hence, my Cyanogen focus is on devices with the Wileyfox Swift.

I will work on getting the Lenovo tablet, but it will be a few weeks before that happens and I can look into this.

In the meantime, you might try building and running the `demo-playground/` app on these devices. You should be able to reproduce your settings using the UI of that app (though it will save the images to its own directory). If the `demo-playground/` app works, then there is something specific with your app that is triggering this problem, and I will not be able to reproduce the issue even if I get hardware.

shahzadyounas, 2016-07-19:
Appreciate that thank you.

Tried the demo playground and got the same result.

Give it a quick go on the emulator - you will see the issue appearing.

commonsguy, 2016-07-23:
The Lenovo S8-50F does not support USB debugging with my development environment. The problem would appear to be on Lenovo's side, as it does not seem to honor the Stay Awake developer setting either, despite the fact that a USB cable is plugged in and the device is charging. I downloaded the `demo-playground/` to the device. I am able to reproduce the problem. Unfortunately, without `adb` access, I have no chance of figuring out what the problem is here.

I cannot create an emulator image with a front-facing camera. And, as I noted previously, the emulator is not a useful test environment for this library.

I will purchase a GT-I9195 (your Galaxy S4 Mini) and will see if the stock ROM exhibits the same problem.

shahzadyounas, 2016-07-24:
Appreciate that thank you.  Awaiting your feedback - otherwise I will likely create my own image preview activity and force rotate the image beforehand to correct it.

commonsguy, 2016-07-25:
@shahzadyounas All that will do is break your app on other devices. The preview is wrong on those test devices and perhaps others. The preview is correct on other devices.

ankur-j9, 2016-07-28:
I am using Galaxy S6-Edge Device, Here also preview is rotated at 90 degree. Same i observed in cam-playground app.

commonsguy, 2016-08-12:
Addressed in v0.6.6.

closed 2016-08-12 by commonsguy

shahzadyounas, 2016-08-15:
Thanks v much - will upgrade and test - thanks for your time and efforts

IzzySoft, 2016-10-13:
Sorry for hijacking this (only temporarily I hope), but as this is about image rotation and you, @commonsguy explicitly mention your "focus is on devices like the Wileyfox Swift": I'm no dev, so I'm not using your API. But I've noticed that all camera apps I've tried (including the stock camera shipping with the Swift) always set "Orientation:0" and instead rotate the image itself – which really is a big annoyance to me (I never had any device or camera doing that). Do you have any clue on how I (as an end-user) can fix that? Thanks a lot in advance for any hints!