commonsguy/cwac-cam2 #244: CameraTwoEngine CountDownLatch

MiguelLavigne, 2016-08-02:
Hi,

First off I would just like to thank the author for this library.  It's well written and easy to work with.  Well appreciated!

I have a small question regarding the `CameraTwoEngine` class.

In the [CameraTwoEngine](https://github.com/commonsguy/cwac-cam2/blob/master/cam2/src/main/java/com/commonsware/cwac/cam2/CameraTwoEngine.java#L246) a CountDownLatch latch is used which I'm assuming to prevent some sort of race condition that some phones may have experienced.

This latch blocks the main thread for 2 seconds each time the CameraFragment is stopped on all phones you've currently white-listed for the v2 api.

If this latch is necessary I was wondering if you could shed some light on the original issue?

I'm asking because I would like to use the CameraFragment in a use case where it needs to be swapped in and out but the delay is causing the UI to block when swapping out.

commonsguy, 2016-08-02:
> In the CameraTwoEngine a CountDownLatch latch is used which I'm assuming to prevent some sort of race condition that some phones may have experienced.

That's modelled after Google's Camera2 code.

> This latch blocks the main thread for 2 seconds each time the CameraFragment is stopped 

It *can* block for up to two seconds. Whether it does or not depends on the behavior of the other threads.

> you could shed some light on the original issue?

My assumption is that they want to avoid closing the camera until the preview transaction is completed. Again, I have modelled this after Google's code.

I will look into trying to improve this, though I cannot make any promises.

MiguelLavigne, 2016-08-02:
Hi Mark,

Thank you for the explanation, for now I can use the `ClassicCameraEngine` instead.

But something is curious because the only way I could get the preview transaction to complete was after `CameraDevice.close()` was called.  And in the current google inspired code the `CameraDevice.close()` is actually called only after the latch is acquired.

Based on my limited knowledge of the camera code to me this means that the preview will never be able to finish (call `onClosed`) until the latch times out.  Again there might other ways to trigger the preview to complete without a call to the `CameraTwoEngine:close()` method which I suppose might render my analysis invalid but I haven't found one in my testing.

If you need some sample code to help reproduce this issue don't hesitate to ask.

Thank you!

commonsguy, 2016-11-20:
Addressed in v0.7.0

closed 2016-11-20 by commonsguy