commonsguy/cwac-cam2 #246: Null camera

marcin-adamczewski, 2016-08-05:
Hello,

We discovered some crashes on our production app. Here is the stack trace:

> Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference
>        at com.commonsware.cwac.cam2.ClassicCameraEngine$1.run(ClassicCameraEngine.java:85)
>        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
>        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
>        at java.lang.Thread.run(Thread.java:818)

It happened on following devices:
- Sony D6503
- Infinix-X600-LTE
- Infinix-X552
- Prestigo PSP7557
- BLU LIFE MARK

OS Versions:
4.x and 5.1

Actually there is a check in onCreate method if device has camera so it should not be a problem. And also camera switch option is hidden when there is only one camera available.

commonsguy, 2016-08-07:
You would appear to be using some earlier version of the library. Line 85 of `ClassicCameraEngine.java` is a closing brace.

I have updated `ClassicCameraEngine` to check for a `null` response to `getParameters()`, skipping that camera. I have no idea if this will help you, as I do not have any of these devices, but it is unlikely to hurt. I will release an update to the library tomorrow that includes this change.

marcin-adamczewski, 2016-08-08:
Hello,

Thanks for quick answer.
We use 0.5.3 version and I can see it is quite old version. I forgot to check release notes before creating an issue, sorry for that. 
We also don't have any of those devices so I hope this fix will help to resolve the issue.

commonsguy, 2016-08-08:
Addressed in v0.6.5.

closed 2016-08-08 by commonsguy