commonsguy/cwac-cam2 #248: CWAC CAM2 0.6.3 and Android API 19

ezesolares, 2016-08-16:
I had been testing the my app on different android apis (i dev with a nexus 6p), started with 19 and it seems that it having issues... i tried to find if the error was on my end but well, seems like its not.
i tested this on the android x86 emulator 800x600 API 19 and the demo app (from git)
When i try to capture, i get this on the android monitor

> 08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 I/dalvikvm: Could not find method com.commonsware.cwac.cam2.AbstractCameraActivity.checkSelfPermission, referenced from method com.commonsware.cwac.cam2.AbstractCameraActivity.hasPermission
08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve virtual method 797: Lcom/commonsware/cwac/cam2/AbstractCameraActivity;.checkSelfPermission (Ljava/lang/String;)I
08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x6e at 0x0007
08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 I/dalvikvm: Could not find method android.app.ActionBar.setElevation, referenced from method com.commonsware.cwac.cam2.AbstractCameraActivity.onCreate
08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve virtual method 21: Landroid/app/ActionBar;.setElevation (F)V
08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x6e at 0x0026
08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 I/dalvikvm: Could not find method com.commonsware.cwac.cam2.AbstractCameraActivity.requestPermissions, referenced from method com.commonsware.cwac.cam2.AbstractCameraActivity.onCreate
08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve virtual method 822: Lcom/commonsware/cwac/cam2/AbstractCameraActivity;.requestPermissions ([Ljava/lang/String;I)V
08-16 17:21:39.262 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x6e at 0x0066
08-16 17:21:39.292 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 I/dalvikvm: Could not find method android.widget.Chronometer.setCountDown, referenced from method com.commonsware.cwac.cam2.CameraFragment.configureChronometer
08-16 17:21:39.292 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve virtual method 418: Landroid/widget/Chronometer;.setCountDown (Z)V
08-16 17:21:39.292 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x6e at 0x003f
08-16 17:21:39.292 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 I/dalvikvm: Could not find method android.widget.ImageButton.<init>, referenced from method com.github.clans.fab.FloatingActionButton.<init>
08-16 17:21:39.292 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve direct method 423: Landroid/widget/ImageButton;.<init> (Landroid/content/Context;Landroid/util/AttributeSet;II)V
08-16 17:21:39.292 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x70 at 0x0003
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: Unable to resolve superclass of Lcom/github/clans/fab/FloatingActionButton$1; (166)
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: Link of class 'Lcom/github/clans/fab/FloatingActionButton$1;' failed
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 E/dalvikvm: Could not find class 'android.graphics.drawable.RippleDrawable', referenced from method com.github.clans.fab.FloatingActionButton.createFillDrawable
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve new-instance 72 (Landroid/graphics/drawable/RippleDrawable;) in Lcom/github/clans/fab/FloatingActionButton;
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x22 at 0x0038
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 E/dalvikvm: Could not find class 'android.graphics.drawable.RippleDrawable', referenced from method com.github.clans.fab.FloatingActionButton.onActionDown
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve check-cast 72 (Landroid/graphics/drawable/RippleDrawable;) in Lcom/github/clans/fab/FloatingActionButton;
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x1f at 0x001d
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 E/dalvikvm: Could not find class 'android.graphics.drawable.RippleDrawable', referenced from method com.github.clans.fab.FloatingActionButton.onActionUp
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve check-cast 72 (Landroid/graphics/drawable/RippleDrawable;) in Lcom/github/clans/fab/FloatingActionButton;
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x1f at 0x001f
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 I/dalvikvm: Could not find method android.widget.ImageButton.setElevation, referenced from method com.github.clans.fab.FloatingActionButton.setElevation
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve virtual method 429: Landroid/widget/ImageButton;.setElevation (F)V
08-16 17:21:39.302 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x6f at 0x000b
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 I/dalvikvm: Could not find method android.widget.ImageButton.setElevation, referenced from method com.github.clans.fab.FloatingActionButton.setElevationCompat
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: VFY: unable to resolve virtual method 429: Landroid/widget/ImageButton;.setElevation (F)V
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: VFY: replacing opcode 0x6f at 0x0023
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: Unable to resolve superclass of Lcom/github/clans/fab/FloatingActionButton$1; (166)
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: Link of class 'Lcom/github/clans/fab/FloatingActionButton$1;' failed
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: DexOpt: unable to opt direct call 0x00c1 at 0x4c in Lcom/github/clans/fab/FloatingActionButton;.createFillDrawable
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: Unable to resolve superclass of Lcom/github/clans/fab/FloatingActionButton$1; (166)
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 W/dalvikvm: Link of class 'Lcom/github/clans/fab/FloatingActionButton$1;' failed
08-16 17:21:39.312 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: DexOpt: unable to opt direct call 0x066e at 0x51 in Lcom/github/clans/fab/FloatingActionButton;.createFillDrawable
08-16 17:21:39.322 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/dalvikvm: GC_FOR_ALLOC freed 193K, 10% free 3043K/3360K, paused 2ms, total 3ms
                                                                                    
                                                                                    [ 08-16 17:21:39.342 29852:29852 D/         ]
                                                                                    HostConnection::get() New Host Connection established 0xb8567090, tid 29852
08-16 17:21:39.382 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/OpenGLRenderer: Enabling debug mode 0
08-16 17:21:39.512 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/Event: No subscribers registered for event class com.commonsware.cwac.cam2.CameraController$ControllerDestroyedEvent
08-16 17:21:39.512 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/Event: No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent
08-16 17:21:41.522 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/Event: No subscribers registered for event class com.commonsware.cwac.cam2.CameraController$ControllerDestroyedEvent
08-16 17:21:41.522 29852-29852/com.commonsware.cwac.cam2.demo:cwac_cam2 D/Event: No subscribers registered for event class de.greenrobot.event.NoSubscriberEvent

commonsguy, 2016-08-16:
Those are fairly typical messages. They do not indicate a problem with the library or your app.

If you have additional concerns, please [follow the instructions for getting support](https://github.com/commonsguy/cwac-cam2#questions).

closed 2016-08-16 by commonsguy

ezesolares, 2016-08-16:
But the activity never opens, it closes as soon as it start and goes back to the activity that called it

commonsguy, 2016-08-16:
That does not change the fact that the messages in your issue do not indicate a problem.

If you feel that there is a problem in the library, please follow [the instructions for filing a bug report](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports).