commonsguy/cwac-cam2 #250: Delay after capture button clicked.

mhassanreza, 2016-08-19:
I am getting issue in almost all devices except nexus devices. when I click on capture button it almost take 4,5 seconds to capture image.

Here are logs which are causing delays. can you let me know how to fix that issue..?

Thanks in advance.

### D/dalvikvm: GC_FOR_ALLOC freed 2959K, 21% free 18966K/23960K, paused 165ms, total 165ms
D/dalvikvm: GC_FOR_ALLOC freed 1441K, 21% free 19137K/23960K, paused 112ms, total 112ms
I/Choreographer: Skipped 32 frames!  The application may be doing too much work on its main thread.
D/dalvikvm: GC_FOR_ALLOC freed 2316K, 22% free 18817K/23960K, paused 104ms, total 105ms
D/dalvikvm: GC_FOR_ALLOC freed 2087K, 22% free 18755K/23960K, paused 98ms, total 99ms
D/dalvikvm: GC_FOR_ALLOC freed 1978K, 22% free 18755K/23960K, paused 101ms, total 102ms
D/dalvikvm: GC_FOR_ALLOC freed 1968K, 22% free 18822K/23960K, paused 117ms, total 118ms
D/dalvikvm: GC_FOR_ALLOC freed 1849K, 22% free 18916K/23960K, paused 99ms, total 100ms
D/dalvikvm: GC_FOR_ALLOC freed 1967K, 22% free 18877K/23960K, paused 126ms, total 127ms
D/dalvikvm: GC_FOR_ALLOC freed 1953K, 21% free 18937K/23960K, paused 109ms, total 110ms
D/dalvikvm: GC_FOR_ALLOC freed 2243K, 22% free 18742K/23960K, paused 517ms, total 520ms

commonsguy, 2016-08-19:
Please follow [the instructions](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports) for filing bug reports.

mhassanreza, 2016-08-19:
Build.MANUFACTURER = Samsung 
Build.PRODUCT = S5 SM-G900I

mhassanreza, 2016-08-19:
Build.MANUFACTURER = Samsung 
Build.PRODUCT = S5 SM-G900I

Unfortunately unable to upload sample project but this issue happen in your demo project you can check it.

Issue is same as described above.

commonsguy, 2016-08-19:
Thanks for the details.

You might try [putting the `CameraActivity` in a separate process](https://github.com/commonsguy/cwac-cam2/blob/master/docs/CameraActivity.md#configuring-the-manifest-entry). You might also try `skipOrientationNormalization()`, which means that if the camera hardware sets the EXIF `Orientation` header, we leave it alone rather than rotate the image to try to remove the need for the header.

mhassanreza, 2016-08-22:
It works. Now delays reduced from 4 second to 57ms.

I thinks issue was: Most phone cameras are landscape, meaning if you take the photo in portrait, the resulting photos will be rotated 90 degrees. In this case, the camera software should populate the EXIF data with the orientation that the photo should be viewed in.

so skipOrientationNormalization() did worked.

commonsguy, 2016-08-22:
> delays reduced from 4 second to 57ms.

Normalizing the orientation should not take anywhere near that amount of time. I'll see what I can do.

commonsguy, 2016-09-10:
I cannot reproduce the problem. I purchased one of these devices, and taking a picture using `demo-playground/` works quickly, with both front-facing and rear-facing cameras, and in both portrait and landscape orientations.

If you can provide instructions for reproducing the issue, please post them.

In the interim, I an closing this issue as not reproducible.

closed 2016-09-10 by commonsguy