commonsguy/cwac-cam2 #253: Crash in CameraActivity.java

aznSlayer, 2016-08-31:
Library version: com.commonsware.cwac:cam2:0.6.6'

# OS Version: 5.0.2
# Device: SM-A500Y
# RAM Free: 28.6%
# Disk Free: 57.5%

#0. Crashed: main: 0 0 0x0000000000000000
       at android.app.FragmentManagerImpl.enqueueAction(FragmentManager.java:1365)
       at android.app.BackStackRecord.commitInternal(BackStackRecord.java:728)
       at android.app.BackStackRecord.commit(BackStackRecord.java:704)
       at com.commonsware.cwac.cam2.CameraActivity.removeFragments(CameraActivity.java:233)
       at com.commonsware.cwac.cam2.CameraActivity.access$000(CameraActivity.java:31)
       at com.commonsware.cwac.cam2.CameraActivity$2.run(CameraActivity.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:145)
       at android.app.ActivityThread.main(ActivityThread.java:6145)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)

--

Fatal Exception: java.lang.IllegalStateException: Activity has been destroyed
       at android.app.FragmentManagerImpl.enqueueAction(FragmentManager.java:1365)
       at android.app.BackStackRecord.commitInternal(BackStackRecord.java:728)
       at android.app.BackStackRecord.commit(BackStackRecord.java:704)
       at com.commonsware.cwac.cam2.CameraActivity.removeFragments(CameraActivity.java:233)
       at com.commonsware.cwac.cam2.CameraActivity.access$000(CameraActivity.java:31)
       at com.commonsware.cwac.cam2.CameraActivity$2.run(CameraActivity.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:145)
       at android.app.ActivityThread.main(ActivityThread.java:6145)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)

#0. Crashed: main: 0 0 0x0000000000000000
       at android.app.FragmentManagerImpl.enqueueAction(FragmentManager.java:1365)
       at android.app.BackStackRecord.commitInternal(BackStackRecord.java:728)
       at android.app.BackStackRecord.commit(BackStackRecord.java:704)
       at com.commonsware.cwac.cam2.CameraActivity.removeFragments(CameraActivity.java:233)
       at com.commonsware.cwac.cam2.CameraActivity.access$000(CameraActivity.java:31)
       at com.commonsware.cwac.cam2.CameraActivity$2.run(CameraActivity.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:145)
       at android.app.ActivityThread.main(ActivityThread.java:6145)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)

commonsguy, 2016-08-31:
Quoting [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md), I need:

- the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish)

- the steps necessary, using that code and that hardware, to reproduce the problem

aznSlayer, 2016-08-31:
Sorry about that, I'll try to write some code to reproduce it!

commonsguy, 2016-09-24:
Closing as not reproducible.

closed 2016-09-24 by commonsguy