commonsguy/cwac-cam2 #256: OutOfMemoryError when taking a picture.

op-and, 2016-09-22:
I am getting an OutOfMemory error when taking a picture on a Samsung S5. The picture shows up in landscape mode even though I am taking the picture in portrait mode. I am using v0.6.7 and was getting the same error with 0.6.6. Thanks.

launch camera:

`    private void launchCameraModule() {
        mCurrentPhotoFile = new File(getActivity().getFilesDir(), FILE_NAME);
        Intent intent = new com.commonsware.cwac.cam2.CameraActivity.IntentBuilder(getActivity())
                .skipConfirm()
                .facing(Facing.BACK)
                .to(mCurrentPhotoFile)
                .zoomStyle(ZoomStyle.SEEKBAR)
                .build();

        startActivityForResult(intent, CAMERA_WITH_DATA);
    }`

Log:

> 09-22 16:14:50.472 15431-16034/com.example.test E/dalvikvm-heap: Out of memory on a 63489040-byte allocation.
09-22 16:14:50.472 15431-16034/com.example.test I/dalvikvm: "pool-12-thread-1" prio=5 tid=27 RUNNABLE
09-22 16:14:50.472 15431-16034/com.example.test I/dalvikvm:   | group="main" sCount=0 dsCount=0 obj=0x42be6580 self=0x7a2e3388
09-22 16:14:50.472 15431-16034/com.example.test I/dalvikvm:   | sysTid=16034 nice=0 sched=0/0 cgrp=apps handle=2049513584
09-22 16:14:50.472 15431-16034/com.example.test I/dalvikvm:   | state=R schedstat=( 432203298 133859771 531 ) utm=30 stm=13 core=1
09-22 16:14:50.472 15431-16034/com.example.test I/dalvikvm:     at android.graphics.Bitmap.nativeCreate(Native Method)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at android.graphics.Bitmap.createBitmap(Bitmap.java:924)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at android.graphics.Bitmap.createBitmap(Bitmap.java:901)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at android.graphics.Bitmap.createBitmap(Bitmap.java:833)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at com.commonsware.cwac.cam2.ImageContext.rotateViaMatrix(ImageContext.java:251)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at com.commonsware.cwac.cam2.ImageContext.getJpeg(ImageContext.java:104)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at com.commonsware.cwac.cam2.JPEGWriter.process(JPEGWriter.java:84)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at com.commonsware.cwac.cam2.PictureTransaction.process(PictureTransaction.java:38)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:470)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
09-22 16:14:50.482 15431-16034/com.example.test I/dalvikvm:     at java.lang.Thread.run(Thread.java:841)
09-22 16:14:50.482 15431-16034/com.example.test E/CWAC-Cam2: Exception in camera processing
                                                                        java.lang.OutOfMemoryError
                                                                            at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                            at android.graphics.Bitmap.createBitmap(Bitmap.java:924)
                                                                            at android.graphics.Bitmap.createBitmap(Bitmap.java:901)
                                                                            at android.graphics.Bitmap.createBitmap(Bitmap.java:833)
                                                                            at com.commonsware.cwac.cam2.ImageContext.rotateViaMatrix(ImageContext.java:251)
                                                                            at com.commonsware.cwac.cam2.ImageContext.getJpeg(ImageContext.java:104)
                                                                            at com.commonsware.cwac.cam2.JPEGWriter.process(JPEGWriter.java:84)
                                                                            at com.commonsware.cwac.cam2.PictureTransaction.process(PictureTransaction.java:38)
                                                                            at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:470)
                                                                            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                                            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                                            at java.lang.Thread.run(Thread.java:841)

commonsguy, 2016-09-22:
This is an issue tracker. It is not a support board. Please [follow the instructions for getting support](https://github.com/commonsguy/cwac-cam2#questions).

[Add the `android:process` attribute to your manifest](https://github.com/commonsguy/cwac-cam2/blob/master/docs/CameraActivity.md#configuring-the-manifest-entry). If that does not help, call `skipOrientationNormalization()` on the `IntentBuilder` to skip changing the image orientation based on the EXIF headers.

closed 2016-09-22 by commonsguy

op-and, 2016-09-23:
Thanks for your help. I apologize - should have looked at the support pages first.