commonsguy/cwac-cam2 #258: Exception java.lang.NullPointerException: Attempt to invoke virtual method 'void android.widget.Chronometer.stop()' on a null object reference

mehdimasrour, 2016-09-27:
Hello again.

Using 0.6.8 version of cwac-cam2

Can't take a single picture with my OnePlus 3 (ONEPLUS A3003) running 6.0.1 Android

Code is always the same :

Intent i = new CameraActivity.IntentBuilder(HomeActivity.this)
.to(ImageUtils.getOutputMediaFileUri(ImageUtils.MEDIA_TYPE_IMAGE))
.skipConfirm()
.zoomStyle(ZoomStyle.SEEKBAR)
.updateMediaStore()
.build();
startActivityForResult(i, CAPTURE_IMAGE_ACTIVITY_REQUEST_CODE);

Stack trace :

```
Exception java.lang.NullPointerException: Attempt to invoke virtual method 'void android.widget.Chronometer.stop()' on a null object reference
com.commonsware.cwac.cam2.CameraFragment.onStop (CameraFragment.java:187)
android.app.Fragment.performStop (Fragment.java:2391)
android.app.FragmentManagerImpl.moveToState (FragmentManager.java:1025)
android.app.FragmentManagerImpl.removeFragment (FragmentManager.java:1253)
android.app.BackStackRecord.run (BackStackRecord.java:758)
android.app.FragmentManagerImpl.execPendingActions (FragmentManager.java:1535)
android.app.FragmentManagerImpl$1.run (FragmentManager.java:482)
android.os.Handler.handleCallback (Handler.java:739)
android.os.Handler.dispatchMessage (Handler.java:95)
android.os.Looper.loop (Looper.java:152)
android.app.ActivityThread.main (ActivityThread.java:5507)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:726)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:616)
```

Thank you in advance.

Regards

commonsguy, 2016-09-27:
Try dropping back to 0.6.7 for the time being.

commonsguy, 2016-09-27:
Addressed in v0.6.9.

closed 2016-09-27 by commonsguy