commonsguy/cwac-cam2 #262: Flash turns off before picture is taken on Nexus 5X when using Camera2 API

Emill, 2016-10-03:
On Nexus 5X (Android 7.0 build NRD90S), if I use the demo-playground, select Flash=Auto, Force specific engine=Camera2 and take a photo in a dark place, the flash turns on but is being turned off before the picture is actually taken, which obviously leads to a very dark image. If I use the old camera engine the picture is taken correctly while the flash is on. I tested some other phones (Samsung S5 and Nexus 6) and it works better there.

commonsguy, 2016-10-13:
I do not control the timing of the flash. I simply request the flash mode. It is up to the camera hardware and Android to control the timing of the flash. Hence, I fail to see where there is anything that I can do about this -- sorry!

closed 2016-10-13 by commonsguy

commonsguy, 2016-10-29:
Oddly, I get this phenomenon with the classic API, and the `camera2` API is behaving better. I am switching the default based on my tests for the next library update.