commonsguy/cwac-cam2 #267: Photo triggering very slow

koeckc, 2016-11-07:
We use version 0.6.9 of the lib and we are experiencing very slow photo triggering on a Samsung device. (Samsung Galaxy S5 Neo, SM-G903F ) at Android SDK: 23 (6.0.1).

After clicking the take photo button it takes very long (10 to 15 secs) until photo is actually taken.

Thats the code we use for building the intent :
`              CameraActivity.IntentBuilder b = new CameraActivity.IntentBuilder(getActivity());
                b.flashMode(FlashMode.AUTO);
		b.quality(AbstractCameraActivity.Quality.HIGH);
		b.allowSwitchFlashMode();
		b.skipConfirm();
		b.zoomStyle(ZoomStyle.SEEKBAR);
		File file;
		file = new File(getActivity().getCacheDir(), "tempfile.jpg");
		b.to(file);`

Is this a known issue ?
Is there maybe a work around ?

commonsguy, 2016-11-07:
Please read [the documentation for how to get support](https://github.com/commonsguy/cwac-cam2#questions).

You may wish to try `skipOrientationNormalization()`, which will not attempt to rotate the image based on supplied EXIF headers.

closed 2016-11-07 by commonsguy

koeckc, 2016-11-07:
Oh sorry i did not read the how to get support link.
Thx that solved our speed issue