commonsguy/cwac-cam2 #268: Video not playing on ios

mourice, 2016-11-09:
Video captured using this lib are not playing on IOS device. I am using video tag for playing the video. If I capture the video on android then its working fine.

commonsguy, 2016-11-09:
This is an issue tracker, not a support board. Please read the documentation for [how to get support](https://github.com/commonsguy/cwac-cam2#questions). Please also read [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md) for how to submit bug reports, as you did not follow those instructions either.

There is nothing that I am going to do specifically about this. I am simply asking the camera hardware to record a video using `MediaRecorder`. I am not in position to guarantee compatibility with arbitrary programs running on arbitrary operating systems, for videos recorded on arbitrary hardware using arbitrary application code that happens to use this library.

You might wish to track issue #269, where I will look to offer somewhat greater configuration control over the video recording process. You could use those options to see if you can find ones that you prefer.

closed 2016-11-09 by commonsguy

mourice, 2016-11-11:
Hi Mark,
Thanks for the response, I think with grater configuration you mean setting
the more profile parameter such as bitrate,frame width/height then it would
be grate. Will follow the issue #269 you mentioned.

....