commonsguy/cwac-cam2 #270: Front facing camera upside down on the Android 7.1 Beta Nexus 6p

ezesolares, 2016-11-15:
I tried my app on the latest beta 7.1 to check if everything worked ok, well, the front facing camera is upside down... to see if it wasnt my mistake, i tested the cam2 demos and the same happens.
I know its beta but just wanted to let you know so you can check what changed for future versions.

Thanks for your work

commonsguy, 2016-11-15:
I am marking this as pending hardware. I do not have a spare 6P to put on the 7.1 beta, so I will look into this once 7.1 ships for the 6P for real.

commonsguy, 2016-12-17:
I cannot reproduce this problem, now that my 6P has 7.1.1 on it. I do not know if:

- I fixed this by something else over the past month
- Google fixed it before the final release
- I am simply not following the proper steps to reproduce it

I am going to close this as not reproducible. If you are able to reproduce the problem, please comment back with very specific steps for doing so, and I will try them.

closed 2016-12-17 by commonsguy