commonsguy/cwac-cam2 #275: Nexus 5X: Downsampled preview image is pixellated

akirmse, 2016-11-21:
On a Nexus 5X running Android 7.0, take a picture, and the resultant preview image looks pixellated.  See attachment.  I do not see the same problem with the same code on a Nexus 5.  My intent is built as:

        Intent intent = new CameraActivity.IntentBuilder(getActivity())
                .to(photoFile)
                .build();

[image]

commonsguy, 2016-11-22:
I cannot reproduce the problem using the `demo-playground/` app on a Nexus 5X, if I request a high-quality image. I *can* reproduce it if I request a low-quality image, because, well, the image is low-quality. So:

- Add `quality(AbstractCameraActivity.Quality.HIGH)` to your builder and see what happens

- Play around with the `demo-playground/` app and see if you can reproduce it there with some combination of settings

akirmse, 2016-11-22:
Thanks for the pointer to the quality setting.  If high quality isn't the default, that might be worth mentioning somewhere.

The problem is actually the quality of the confirmation image, not necessarily the final image.  I can reproduce the problem in demo-playground on a 5X by turning on "Show confirmation screen", even if "Use high quality" is set, if "Confirmation quality" is "default", 0.01, or 1.0.  Other values for Confirmation quality seem to work OK.  I set Confirmation quality to 0.5 in my app and the confirmation image looks OK now.

commonsguy, 2016-11-22:
> If high quality isn't the default, that might be worth mentioning somewhere.

It is the default and is documented as such. However, when supporting people using this library, I have to grasp at straws.

> The problem is actually the quality of the confirmation image, not necessarily the final image.

Yes, but the starting point for the confirmation image quality is the photo quality. A low-quality photo guarantees a low-quality confirmation image, since the confirmation image *is* the photo, simply loaded into memory.

> I can reproduce the problem in demo-playground on a 5X by turning on "Show confirmation screen", even if "Use high quality" is set, if "Confirmation quality" is "default", 0.01, or 1.0.

I cannot reproduce the problem on a Nexus 5X, running stock Android 7.0, using v0.7.0's `demo-playground/` with a high-quality image. Here is the confirmation screen with confirmation quality set at 0.01:

[image]

Here is the same scene, with a confirmation quality of 1.0:

[image]

Here is the same scene, with a confirmation quality of "default":

[image]

Admittedly, my scene is not the same, as I don't have an ottoman(?) like yours, and my prop is cherry rather than vanilla. But, I would expect at least *something* here, and all the confirmation images seem fine.

That, plus the fact that you have things working fine for you now, is why I am closing this as non-reproducible. Issue #276 may help me deal with this sort of scenario in the future, as perhaps there is something else afoot in my `demo-playground/` settings that differs from yours that is causing me to have more reliable confirmation image resampling. If you have other ideas as to how I might reproduce the problem, let me know.

closed 2016-11-22 by commonsguy