commonsguy/cwac-cam2 #278: NPE com.commonsware.cwac.cam2.CameraTwoEngine$CapturePictureTransaction.unlockFocus (CameraTwoEngine.java:714)

shahzadyounas, 2016-11-23:
Hi

I've updated my app to include the latest build of cam2 - compile 'com.commonsware.cwac:cam2:0.7+'

However I am now getting a new error I was not getting before:

`Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.camera2.CameraCaptureSession.capture(android.hardware.camera2.CaptureRequest, android.hardware.camera2.CameraCaptureSession$CaptureCallback, android.os.Handler)' on a null object reference
       at com.commonsware.cwac.cam2.CameraTwoEngine$CapturePictureTransaction.unlockFocus(CameraTwoEngine.java:714)
       at com.commonsware.cwac.cam2.CameraTwoEngine$CapturePictureTransaction.onCaptureCompleted(CameraTwoEngine.java:696)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
       at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:211)
       at android.os.HandlerThread.run(HandlerThread.java:61)`

Across a range of devices, Samsung, Sony, Motorola:
(eg samsung SM-G920F, GT-I9505, SM-G930F)

Note - was working fine in previous builds of cam2.

Thanks in advance!

commonsguy, 2016-11-23:
Yes, I ran into that too on some devices. It's tied into my fix for issue #244. I plan to release a 0.7.1 this weekend that contains a fix. In the meantime, stick with the 0.6.x series.

shahzadyounas, 2016-11-23:
Gotcha  - thanks for that - will do.

alecio, 2016-11-25:
we've got it reproducible too!

commonsguy, 2016-11-27:
Addressed in v0.7.1.

closed 2016-11-27 by commonsguy