commonsguy/cwac-cam2 #280: Wrong orientation on landscape

hoanganh200870, 2016-11-25:
I record video on landscape mode then get video with wrong orientation. Please check https://drive.google.com/file/d/0B5a-fsZbLrYoLUVlZmZCMlFqZDA/view?usp=drivesdk. I created by demo app on LGG2 . I also tested with Nexus 5, Zenphone 2.

commonsguy, 2016-11-25:
I do not anticipate being able to obtain an LG-G802 for testing. If and when I do, I will update this issue.

hoanganh200870, 2016-11-25:
So, which device can you accept? I can send zip file for Nexus 5 and Zenfone 2.

commonsguy, 2016-11-25:
Are you saying that you are seeing the same issue with your Nexus 5? If so, I will not be able to reproduce the problem on the Nexus 5. I just tested that with v0.7.0 of the library within the past week, and it works fine. If you are experiencing problems with the Nexus 5, you need to determine why *you* are experiencing problems with your specific Nexus 5. For example, have you changed something in the device settings that might somehow impact orientation behavior?

hoanganh200870, 2016-11-26:
I'm not focusing on specified devices. I'm working to build camera function base on this library. And the result different with my expectation. Maybe we have a gap in understanding.

Please check picture in below. On the left is currect result, and the right is my expectation.

[image]

I think it relates to the closed issue https://github.com/commonsguy/cwac-cam2/issues/56.

commonsguy, 2016-11-26:
Let's go all the way back to the beginning.

> I record video on landscape mode then get video with wrong orientation. Please check https://drive.google.com/file/d/0B5a-fsZbLrYoLUVlZmZCMlFqZDA/view?usp=drivesdk. I created by demo app on LGG2 .

When I unpack your ZIP archive and look at the videos, the two `-landscape` videos appear to be saved with portrait metadata. When I play those videos, my video player rotates the output to portrait. However, the `-portrait` and `-landscape` videos are different, suggesting that you used the `demo/` app properly.

Now, as it turns out, I *do* have this very model of device &mdash; I had not realized that my G2 was a D802 model. The D802 appears to have been sold in the Asia/Pacific region, and I do not have very many devices from there, as that is not where I live.

When I run the `demo/` app on my LG G2 D802, I get correct results:

https://s3.amazonaws.com/misc.commonsware.com/cam2_LGE_g2_open_com_20161126-074648.zip

If you examine that ZIP file, you will see that both `-landscape` videos are recorded in landscape.

There are two differences in our environments:

1. I am using a `demo/` app from a version of the library that is slightly newer than what you have access to, as I will be shipping a v0.7.1 update soon. It is possible, though somewhat unlikely, that the problem exists in your `demo/` app but not mine. If you would like to test with the same `demo/` app version that I just tried, you can download the APK from:

https://s3.amazonaws.com/misc.commonsware.com/demo-debug.apk

2. Your device may be configured differently than mine. Other than applying OS updates (back when there were some) and enabling developer options/USB debugging/etc., I have not changed much of anything on this device.

The key is that I do not have your device. *Usually*, when this sort of problem arises, the problem lies in the hardware, not something the device user may have done. In this case, since I happen to have the same make and model of device, I have to rule out the hardware. And so unless you can come up with an explanation for why your device has a problem and my device does not, there is little that I can do.

> I also tested with Nexus 5, Zenphone 2.

You did not say whether you were experiencing problems with these devices. I originally assumed that those devices were fine.

> So, which device can you accept? I can send zip file for Nexus 5 and Zenfone 2.

Now, you seem to be indicating that you are experiencing the same problem with these two devices.

I have a Nexus 5. The results are the same as with the D802: you are having a problem, and I am not. I do not have a Zenfone 2, but until we determine what is going on with the Nexus 5 and the D802, I am not going to invest in more hardware based on this issue report.

> I'm not focusing on specified devices. 

I am, because you are filing an issue against this library, and this functionality works fine in general.

> I think it relates to the closed issue #56.

I do not think so, as I would expect to be reproducing the problem.

----------

So, try the `demo/` app that I linked to, or wait until v0.7.1 is released, then re-test your devices.

If the devices now work fine, then obviously I broke something, then fixed it. In which case, who-hoo! :-)

If the devices still exhibit the problem, I would need some ideas from you about what is different about your devices compared to mine, given that they are the same device models.

hoanganh200870, 2016-11-29:
I think I know what happens. We have another different in testing step. I still lock rotation all the time. I used previous project https://github.com/commonsguy/cwac-camera and get expected result in this case, but can not in this project.

How do you think about this case? I lock orientation and record video in landscape mode.

commonsguy, 2016-11-29:
If you are referring to locking orientation in your code, use `orientationLockMode()` on the `VideoRecorderActivity.IntentBuilder`. Pass in `OrientationLockMode.LANDSCAPE` as the value.

If you are referring to locking orientation as a user, I would expect problems. If the old project worked successfully, that was purely by accident.

hoanganh200870, 2016-11-29:
Let me explain my case as a user. I open the camera and rotate my phone 90deg (landscape mode). Tap the button to record video. Then I get video with wrong in orientation. I don't know I need to enable orientation to record in landscape mode. I think it's a bug.

I think we need to lock the orientation when user taps the button, not when initializing. How do you think?

commonsguy, 2016-11-29:
> I open the camera and rotate my phone 90deg (landscape mode). Tap the button to record video. Then I get video with wrong in orientation.

Not normally. See [my earlier comment](https://github.com/commonsguy/cwac-cam2/issues/280#issuecomment-263062416).

> I think we need to lock the orientation when user taps the button, not when initializing. How do you think?

That is the default behavior with v0.7.0 and higher.

hoanganh200870, 2016-11-29:
I think it get current orientation, but not correctly. I lock orientation, rotate 90deg but I still detect I'm in portrait. I think correct is landscape. It is right?

commonsguy, 2016-11-29:
If you, as a user, are locking orientation at the OS level, there is nothing that I can do about that, to the best of my knowledge.

hoanganh200870, 2016-11-29:
So, the current version doesn't support this case. In the old library, I can handle by getting current orientation (use sensor) and use `setHintOrientation` function. But in this library, I think it will conflict with orientation plugin. Can you give me a hint for this?

commonsguy, 2016-11-29:
> In the old library, I can handle by getting current orientation (use sensor) and use setHintOrientation function.

This library is already doing that.

hoanganh200870, 2016-11-29:
Let me correct again. If I lock orientation, open the camera, rotate device 90deg then record video. I will got my issue. Is it right?

Please help me check this case. I checked in 4 devices I have.

commonsguy, 2016-11-29:
If, as a user, you switch your device from auto-rotate to portrait, hold the device in landscape, and record the video, you will get results that match your reported issue.

Open Camera appears to be able to handle this case, so I'll look into it sometime.

hoanganh200870, 2016-11-29:
Thank you for intention to enhance. I don't think it will be done soon, but I hope I can see it asap.