commonsguy/cwac-cam2 #287: Confirmation image always rotated on Nexus 5X

akirmse, 2016-12-02:
On a stock Nexus 5X with Android 7.0 and the playground app, if I take a picture looking straight down, the image looks fine until I take the picture, and then it appears rotated 90 degrees counterclockwise on the confirmation screen.  I tried skipOrientationNormalization() and orientationLockMode(OrientationLockMode.PORTRAIT) but the behavior is the same.

I can't reproduce the problem with a Nexus 5.

commonsguy, 2016-12-02:
Does "take a picture looking straight down" mean you are taking a picture of something on the floor?

I am just trying to ensure that I understand your scenario. I am sure that your floor is perfectly lovely and worthy of memorializing in the form of digital photography. :-)

akirmse, 2016-12-02:
Yes indeed, in addition to preserving pictures of soda cans for posterity, I enjoy recording the state of early 21st century flooring.

My actual use case is taking pictures of USGS benchmarks to send to the appropriate government agency.  They require two pictures taken straight down.

commonsguy, 2016-12-02:
I am scratching my head at what "USGS benchmarks" might be, other than, y'know, rocks.

Regardless, I can reproduce the problem. I'll take a look at it, though this sort of thing screams "hardware-specific behavior"...

commonsguy, 2016-12-03:
On the Nexus 5X, I get the behavior that you cite if I start in portrait mode, then rotate the device to be parallel to the ground. If I start in landscape mode, then rotate the device to be parallel to the ground, I get a 180-degree flip. 

The closest fix that I might be able to do would be to track each and every movement of the device, detect when we are getting close to parallel to the ground, and at that point rely on the orientation before we got to the "close to parallel to the ground" state, rather than whatever the sensors are telling me at that point. This will not be a 100% solution, most likely, though it probably would give more reliable results.

If somebody wants this any time soon, [contact me](mailto:consulting@commonsware.com) regarding a directed development contract. For now, I quite literally cannot afford to spend the time it will take to try to get a semi-reliable solution here.

akirmse, 2016-12-03:
Are you using the camera or camera2 API?  I came across some posts
indicating that the 5X's camera was installed "upside down", and that
camera2 handles this automatically.  I'm seeing a 90 degree rotation in
preview, though, not 180 degrees.

https://www.reddit.com/r/Android/comments/3rjbo8/nexus5x_marshmallow_camera_problem/cwqzqgh/
http://www.androidcentral.com/google-reveals-why-some-photos-taken-nexus-5x-are-upside-down

commonsguy, 2016-12-03:
Actually, I am seeing the exact opposite of what you're describing.

With the camera2 API:
- in quasi-portrait, the image is off by 90 degrees
- in quasi-landscape, the image is off by 180 degrees

(here, "quasi-" meaning the orientation of the activity, where the device is parallel to the ground)

With the classic `camera` API, though, both scenarios work fine, with correct preview output.

If you could, use `demo-playground/` and force the engine to be the classic engine, and see if you get the same findings. Flipping the Nexus 5X to use the classic engine is trivial, and if that's a viable workaround for this particular bit of hardware, that would be wonderful. I'd then plan on adding horizontal-plane tests for my next full pass through my hardware, to see how other devices behave.

akirmse, 2016-12-04:
I see the same here.  In a few days I'll try setting the engine to classic
for just the Nexus 5X, as that seems likely to fix the problem.

Is is possible your library is doing some rotation compensation that isn't
necessary with camera2, which apparently camera2 compensates for itself,
causing problems with the 5X's rotated hardware?

commonsguy, 2016-12-04:
> Is is possible your library is doing some rotation compensation that isn't necessary with camera2, which apparently camera2 compensates for itself, causing problems with the 5X's rotated hardware?

I am not doing anything specific for when the device is parallel to the ground, and for more traditional orientations, my `CameraTooEngine` holds up just fine. So, I have to assume that what I'm doing is basically correct, otherwise I would expect to have problems with portrait and landscape.

akirmse, 2016-12-06:
Forcing the engine to classic on the 5X does solve the problem, so maybe
this can be closed.  But it does seem fishy that the old engine works,
while camera2, which is supposed to compensate for the rotated sensor,
doesn't.  I suspect there's still something off in the code.

commonsguy, 2016-12-06:
> Forcing the engine to classic on the 5X does solve the problem, so maybe this can be closed

I'm going to leave it open. I'll update the default configuration for the 5X in the next patch release, and I'll test the horizontal-plane scenario across my device fleet when I work on 0.8.0.

aphexcx, 2017-01-10:
Chiming in to say that this exact issue is happening on a Nexus 6P for me. I'm running 7.1.1 if that matters. skipOrientationNormalization() and orientationLockMode(OrientationLockMode.PORTRAIT) don't help. Any workarounds I should try?

commonsguy, 2017-01-10:
@aphexcx:

> Interestingly, if I force engine to CLASSIC, the image comes out with the correct orientation.

Yes, that was the tactical solution for the 5X.

> Any workarounds I should try?

None that I know of, other than using the classic engine.

> Forcing the engine to classic isn't a great workaround to me because I require flash (enabled via FlashMode.Always), and with the classic engine, photos seem to 'miss' the flash (they come out really dark).

Since AFAIK I have no control over the flash timing, there is nothing I can really do about that.

HelenaSewell, 2017-05-12:
I'm seeing this too, on a Motorola Moto G (2nd gen). Android version 6.0. The issue was present on both mine and the demo playground apps. The effects were pretty much exactly as described above - pointing downwards opens the camera in portrait but gives a landscape image, and forcing orientation to portrait or skipping normalization has no effect. (If I take regular landscape photos using a portrait orientation camera with those settings, that photo is landscape too). 
I didn't see the "flip 180" issue you mentioned above. 
Also, I tried all these things with orientation mode fixed to landscape and everything worked the way it should.
Again, forcing Classic showed none of these problems.
I tried with front-facing camera (default, not classic) too just to check, and that showed the same problem.