commonsguy/cwac-cam2 #291: sony xperia Z2

SoNep, 2016-12-07:
`12-07 15:00:29.919 27013-27799/com.commonsware.cwac.cam2.playground:cwac_cam2 E/RequestThread-0: Received device exception during capture call: 
                                                                                                 java.lang.RuntimeException: autoFocus failed
                                                                                                     at android.hardware.Camera.native_autoFocus(Native Method)
                                                                                                     at android.hardware.Camera.autoFocus(Camera.java:1299)
                                                                                                     at android.hardware.camera2.legacy.LegacyFocusStateMapper.processRequestTriggers(LegacyFocusStateMapper.java:205)
                                                                                                     at android.hardware.camera2.legacy.RequestThreadManager$5.handleMessage(RequestThreadManager.java:808)
                                                                                                     at android.os.Handler.dispatchMessage(Handler.java:98)
                                                                                                     at android.os.Looper.loop(Looper.java:211)
                                                                                                     at android.os.HandlerThread.run(HandlerThread.java:61)`

commonsguy, 2016-12-07:
Per [the repo's contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports), please provide:

- the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem
- the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish)
- the steps necessary, using that code and that hardware, to reproduce the problem

commonsguy, 2016-12-17:
Closing as not reproducible, due to lack of detail. If you comment back with the details, I can reopen the issue.

closed 2016-12-17 by commonsguy

VBegoyan, 2017-11-10:
I have experienced with this issue on camera2. Device is Sony D6503. Android 6.0.1, Api 23
Received device exception during capture call: 
                                                                     java.lang.RuntimeException: autoFocus failed
                                                                         at android.hardware.Camera.native_autoFocus(Native Method)
                                                                         at android.hardware.Camera.autoFocus(Camera.java:1300)
                                                                         at android.hardware.camera2.legacy.LegacyFocusStateMapper.processRequestTriggers(LegacyFocusStateMapper.java:205)
                                                                         at android.hardware.camera2.legacy.RequestThreadManager$5.handleMessage(RequestThreadManager.java:824)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:98)
                                                                         at android.os.Looper.loop(Looper.java:234)
                                                                         at android.os.HandlerThread.run(HandlerThread.java:61)