commonsguy/cwac-cam2 #293: Emulator issues

shahzadyounas, 2016-12-09:
Hi

I know we don't aim for the camera to work in the emulator - but < 0.7 it definitely worked fine. But now on Android API 23 onwards, it shows the surface (eg with emulated camera) but then when taking a photo, does nothing (does not return result to calling activity).

Also on API 22 the surface just doesn't load:

```                                                                         HostConnection::get() New Host Connection established 0xa1169340, tid 4316
12-09 17:14:36.708 4147-4316/com.muzmatch.muzmatchapp E/EGL_emulation: tid 4316: eglCreateContext(887): error 0x3005 (EGL_BAD_CONFIG)
12-09 17:14:36.708 4147-4316/com.muzmatch.muzmatchapp E/CameraDeviceGLThread-1: Received exception on GL render thread: 
                                                                                java.lang.IllegalStateException: eglCreateContext: EGL error: 0x3005
                                                                                    at android.hardware.camera2.legacy.SurfaceTextureRenderer.checkEglError(SurfaceTextureRenderer.java:530)
                                                                                    at android.hardware.camera2.legacy.SurfaceTextureRenderer.configureEGLContext(SurfaceTextureRenderer.java:443)
                                                                                    at android.hardware.camera2.legacy.SurfaceTextureRenderer.configureSurfaces(SurfaceTextureRenderer.java:645)
                                                                                    at android.hardware.camera2.legacy.GLThreadManager$1.handleMessage(GLThreadManager.java:89)
                                                                                    at android.os.Handler.dispatchMessage(Handler.java:98)
                                                                                    at android.os.Looper.loop(Looper.java:135)
                                                                                    at android.os.HandlerThread.run(HandlerThread.java:61)```

Any thoughts/ideas?

commonsguy, 2016-12-09:
Try using the classic engine instead of `camera2`.

shahzadyounas, 2016-12-09:
BTW not sure if this helps - when you tap the take photo...this is the log:

`12-09 17:34:54.907 1199-1596/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_ZOOM
12-09 17:34:54.907 1199-1596/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_COMPRESSED_IMAGE
12-09 17:34:54.907 1199-1596/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_PREVIEW_METADATA
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Camera: takePicture
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Camera: doStopPreview
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Device: stopDeliveringFrames
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Device: stopWorkerThread
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Device: Stopping emulated camera device's worker thread...
12-09 17:34:54.908 1199-3017/? V/EmulatedCamera_Device: Select: THREAD_STOP message is received
12-09 17:34:54.908 1199-3017/? V/EmulatedCamera_FakeDevice: inWorkerThread: Worker thread has been terminated.
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Device: Emulated camera device's worker thread has been stopped.
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_FakeDevice: stopDevice
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Preview: stopPreview
12-09 17:34:54.908 1199-1596/? D/EmulatedCamera_Camera: Starting camera for picture: NV21(jpeg)[640x480]
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_FakeDevice: startDevice
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Device: commonStartDevice: Allocated 0xb3600000 460800 bytes for 307200 pixels in NV21[640x480] frame
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Device: startDeliveringFrames
12-09 17:34:54.908 1199-1596/? V/EmulatedCamera_Device: startWorkerThread
12-09 17:34:54.909 1199-3026/? V/EmulatedCamera_Device: Starting emulated camera device worker thread...
12-09 17:34:54.909 1199-3026/? V/EmulatedCamera_Device: Emulated device's worker thread has been started.
12-09 17:34:54.931 1199-3026/? V/EmulatedCamera_CallbackNotifier: disableMessage: msg_type = 0x2
12-09 17:34:54.931 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_SHUTTER
12-09 17:34:54.931 1199-3026/? V/EmulatedCamera_CallbackNotifier: **** Currently enabled messages:
12-09 17:34:54.931 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_ERROR
12-09 17:34:54.931 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_FOCUS
12-09 17:34:54.931 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_ZOOM
12-09 17:34:54.931 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_COMPRESSED_IMAGE
12-09 17:34:54.931 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_PREVIEW_METADATA
12-09 17:34:54.931 1199-3026/? D/skia: onFlyCompress
12-09 17:34:54.934 1199-3026/? V/EmulatedCamera_JPEGStub: JpegStub_compress: Compressed JPEG: 460800[640x480] -> 21726 bytes
12-09 17:34:54.935 1199-3026/? V/EmulatedCamera_CallbackNotifier: disableMessage: msg_type = 0x100
12-09 17:34:54.935 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_COMPRESSED_IMAGE
12-09 17:34:54.935 1199-3026/? V/EmulatedCamera_CallbackNotifier: **** Currently enabled messages:
12-09 17:34:54.935 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_ERROR
12-09 17:34:54.935 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_FOCUS
12-09 17:34:54.935 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_ZOOM
12-09 17:34:54.935 1199-3026/? V/EmulatedCamera_CallbackNotifier:     CAMERA_MSG_PREVIEW_METADATA
12-09 17:34:54.936 2810-3007/com.muzmatch.muzmatchapp I/RequestThread-1: Received jpeg.
12-09 17:34:54.936 2810-3007/com.muzmatch.muzmatchapp I/RequestThread-1: Producing jpeg buffer...
12-09 17:34:54.936 2810-3007/com.muzmatch.muzmatchapp E/gralloc_ranchu: gralloc_lock usage mismatch usage=0x30 cb->usage=0x3
12-09 17:34:54.936 2810-3007/com.muzmatch.muzmatchapp W/GraphicBufferMapper: lock(...) failed -22 (Invalid argument)
12-09 17:34:54.936 2810-3007/com.muzmatch.muzmatchapp E/Legacy-CameraDevice-JNI: produceFrame: Failed to lock buffer, error Invalid argument (-22).
12-09 17:34:54.936 2810-3007/com.muzmatch.muzmatchapp E/Legacy-CameraDevice-JNI: LegacyCameraDevice_nativeProduceFrame: Error while producing frame Invalid argument (-22).
12-09 17:34:54.936 2810-3007/com.muzmatch.muzmatchapp W/RequestThread-1: Surface abandoned, dropping frame. 
                                                                         android.hardware.camera2.legacy.LegacyExceptionUtils$BufferQueueAbandonedException
                                                                             at android.hardware.camera2.legacy.LegacyExceptionUtils.throwOnError(LegacyExceptionUtils.java:64)
                                                                             at android.hardware.camera2.legacy.LegacyCameraDevice.produceFrame(LegacyCameraDevice.java:636)
                                                                             at android.hardware.camera2.legacy.RequestThreadManager$2.onPictureTaken(RequestThreadManager.java:235)
                                                                             at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1092)
                                                                             at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                             at android.os.Looper.loop(Looper.java:148)
[... cut ...]
12-09 17:34:54.979 2810-3015/com.muzmatch.muzmatchapp W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
12-09 17:34:54.979 2810-3015/com.muzmatch.muzmatchapp W/LegacyMetadataMapper: convertAfModeToLegacy - ignoring unsupported mode 4, defaulting to fixed
12-09 17:34:54.979 2810-3015/com.muzmatch.muzmatchapp W/LegacyRequestMapper: convertRequestToMetadata - Ignoring android.lens.focusDistance false, only 0.0f is supported
12-09 17:34:55.075 1199-1583/? D/AudioFlinger: mixer(0xb4440000) throttle end: throttle time(1)
`

commonsguy, 2017-01-29:
I have no problem with the 7.1 and 7.0 emulators.

On the 6.0 emulator, I have no problem using the classic engine. I get the symptoms that you describe using the Camera2 engine. I have added code to `CameraConstraints` to steer the emulator (or at least the 6.0 one) to the classic engine.

I cannot get the `CameraActivity` to launch on the 5.1 emulator. But, since the emulators have all sorts of other problems, and since zero users use emulators, I cannot afford to spend time chasing my tail trying to figure out why this one emulator image is having difficulty.

commonsguy, 2017-01-29:
Addressed in v0.7.3

closed 2017-01-29 by commonsguy

shahzadyounas, 2017-01-31:
Thanks for investigating - will try with 7+ emulators and see how I get on.

KanUchida, 2017-05-01:
this happens to me as well on 7+ emulators

commonsguy, 2017-05-01:
@KanUchida I need to know *exactly* the steps to take to reproduce the problem. As [I noted previously](https://github.com/commonsguy/cwac-cam2/issues/293#issuecomment-275917580), I was unable to reproduce the problem on 7.0 or 7.1 emulators.