commonsguy/cwac-cam2 #295: LG G4 - Android 6.0

danielgindi, 2016-12-14:
Camera closing immediately after opening

commonsguy, 2016-12-14:
Please [follow the instructions for filing bug reports](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports).

danielgindi, 2016-12-15:
```
Build.MANUFACTURER == "LGE"
Build.PRODUCT == "p1_global_com"
```

danielgindi, 2016-12-15:
This is what shows in the console everytime:

```
12-15 20:51:06.887 3688-4771/? E/CustomThreadFactory: UNCAUGHT in thread DataSyncServiceHelper-1
                                                      java.lang.IllegalStateException: could not connect to GoogleApiClient: ConnectionResult{statusCode=unknown status code 16, resolution=null}
                                                          at com.google.android.wearable.datatransfer.internal.DefaultPeerProvider.blockingConnect(DefaultPeerProvider.java:75)
                                                          at com.google.android.wearable.datatransfer.internal.DefaultPeerProvider.getConnectedPeers(DefaultPeerProvider.java:32)
                                                          at com.google.android.wearable.datatransfer.internal.DataSyncServiceHelper.processIntent(DataSyncServiceHelper.java:378)
                                                          at com.google.android.wearable.datatransfer.internal.DataSyncServiceHelper.access$000(DataSyncServiceHelper.java:50)
                                                          at com.google.android.wearable.datatransfer.internal.DataSyncServiceHelper$IntentProcessor.run(DataSyncServiceHelper.java:719)
                                                          at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                          at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                          at java.lang.Thread.run(Thread.java:818)
12-15 20:51:07.529 4130-5009/? E/DatabaseUtils: Writing exception to parcel
                                                java.lang.SecurityException: Permission Denial: reading com.android.providers.media.MediaProvider uri content://media/external/images/media from pid=9189, uid=10110 requires android.permission.READ_EXTERNAL_STORAGE, or grantUriPermission()
                                                    at android.content.ContentProvider.enforceReadPermissionInner(ContentProvider.java:605)
                                                    at android.content.ContentProvider$Transport.enforceReadPermission(ContentProvider.java:480)
                                                    at android.content.ContentProvider$Transport.query(ContentProvider.java:211)
                                                    at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:112)
                                                    at android.os.Binder.execTransact(Binder.java:453)
12-15 20:51:07.531 9189-9975/? E/iu.UploadsManager: Insufficient permissions to process media
                                                    java.lang.SecurityException: Permission Denial: reading com.android.providers.media.MediaProvider uri content://media/external/images/media from pid=9189, uid=10110 requires android.permission.READ_EXTERNAL_STORAGE, or grantUriPermission()
                                                        at android.os.Parcel.readException(Parcel.java:1599)
                                                        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:183)
                                                        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:135)
                                                        at android.content.ContentProviderProxy.query(ContentProviderNative.java:421)
                                                        at android.content.ContentResolver.query(ContentResolver.java:498)
                                                        at android.content.ContentResolver.query(ContentResolver.java:441)
                                                        at ish.a(PG:110)
                                                        at com.google.android.libraries.social.autobackup.FingerprintScannerIntentService.onHandleIntent(PG:81)
                                                        at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:66)
                                                        at android.os.Handler.dispatchMessage(Handler.java:102)
                                                        at android.os.Looper.loop(Looper.java:148)
                                                        at android.os.HandlerThread.run(HandlerThread.java:61)
```

Seems to have something to do with an embedded fingerprint scanner exposing itself as a camera?

commonsguy, 2016-12-15:
Nothing in those stack traces pertain to this library. For example, there is no reference to `com.commonsware.cwac.cam2` classes.

Also, please note that I still need:

- the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish)
- the steps necessary, using that code and that hardware, to reproduce the problem

danielgindi, 2016-12-15:
I know it doesn't say "cwac", it would be easier that way. But something is going on deeper in.
I'm testing with the demo app right now to see what happens.

danielgindi, 2016-12-15:
It happens with the `demo` too! 
But then the camera opens - closes - and after 1 second - opens again, and then I can take a picture

danielgindi, 2016-12-15:
```{"api":23,"brand":"lge","build":"MRA58K","cpu":["arm64-v8a",""],"hardware":"p1","manufacturer":"LGE","model":"LG-H815","product":"p1_global_com","version":"6.0"}```

danielgindi, 2016-12-15:
Isn't it fun discovering new sh** on every single Android device?

danielgindi, 2016-12-16:
```
    add(new Builder()
            .manufacturer("LGE")
            .product("p1_global_com")
            .supportsCameraTwo(false)
            .build());
```

Solved the problem.

commonsguy, 2016-12-16:
I will still look into getting hardware for this, but this might wind up being addressed by #289. Thanks for the research!

danielgindi, 2017-01-01:
Seems to be the same case with `manufacturer=LGE, product=jagnm_global_com`, which is the LG-G3

commonsguy, 2017-01-01:
@danielgindi I have moved that one to issue #299.

commonsguy, 2017-01-29:
Addressed in v0.7.3

closed 2017-01-29 by commonsguy