commonsguy/cwac-cam2 #299: Block Camera2 on LG G3

commonsguy, 2017-01-01:
`MANUFACTURER` = `LGE`, `PRODUCT` = `jagnm_global_com`, per [this comment](https://github.com/commonsguy/cwac-cam2/issues/295#issuecomment-269901393).

rupinderjeet, 2017-01-03:
It seems that Almalence's OpenCamera project has done something to support LG G- devices. If you haven't looked, [check CameraController class here](https://github.com/almalence/OpenCamera/blob/master/src/com/almalence/opencam/cameracontroller/CameraController.java).

danielgindi, 2017-01-23:
On LG G3 I see a problem even when manually blocking the Camera2.

```
22:25:36 W System.err : android.content.pm.PackageManager$NameNotFoundException: com.my.app:cwac_cam2
22:25:36 W System.err : at android.app.ApplicationPackageManager.getApplicationInfo(ApplicationPackageManager.java:306)
22:25:36 W System.err : at com.lge.eodengine.EodDetector.changeTopActivity(EodDetector.java:81)
22:25:36 W System.err : at com.lge.eodengine.EodEngineService$1.onForegroundActivitiesChanged(EodEngineService.java:166)
22:25:36 W System.err : at android.app.IProcessObserver$Stub.onTransact(IProcessObserver.java:55)
22:25:36 W System.err : at android.os.Binder.execTransact(Binder.java:453)
```

It does not happen on any other device. But consistently on this damn phone.
And then the camera stays black.

I'm trying to investigate the problem, to see why the heck would it look for that package name containing `:cwac_cam2`, or why would it fail.

commonsguy, 2017-01-23:
`:cwac_cam2` looks like a process name. That is what I name [the process used by the library in the `demo-playground` project](https://github.com/commonsguy/cwac-cam2/blob/v0.7.2/demo-playground/src/main/AndroidManifest.xml#L35-L43). Why LG would be attempting to find a package based on a process name, or what the heck `changeTopActivity()` is trying to do, I cannot say.

danielgindi, 2017-01-25:
Okay so I've copied over these from OpenCamera, and it works:

```
boolean isG4 = Build.MANUFACTURER.equals("LGE") && (
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-h818")  ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-h815")  ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-h812")  ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-h810")  ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-h811")  ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-ls991") ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-vs986") ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-us991")
  );

boolean isG3 = Build.MANUFACTURER.equals("LGE") && (
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-d85")   ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-d72")   ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-d69")   ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-vs985") ||
    Build.MODEL.toLowerCase(Locale.US).startsWith("lg-ls990")
  );
```

Seems like there are many many sub-models on those devices, and they all cause the same trouble.

commonsguy, 2017-01-25:
I guess that I do not understand what these lines are for.

danielgindi, 2017-01-26:
These are to detect all submodels of G4 and G3, which are all causing trouble with Camera2. (Apparantly in OpenCamera they also reverted to the old API for these, but then they found that they need to increase latency or something like that).

Any way if we use such detection - we cache them all. It's using `startsWith` because there are additional digits/letters possible.

commonsguy, 2017-01-29:
I cannot use OpenCamera code, as OpenCamera is GPLv3. I have settled for having all LG phones use the classic engine.

commonsguy, 2017-01-29:
Addressed in v0.7.3

closed 2017-01-29 by commonsguy

rupinderjeet, 2017-03-07:
As @danielgindi said, if you want to find out device names, you can use this library https://github.com/jaredrummler/AndroidDeviceNames just found it today.