commonsguy/cwac-cam2 #303: Camera freezing up the app after clicking camera on latest(0.7.2) version of cam2.

abhiank209, 2017-01-15:
Hello, 

This is the process I followed which caused the error.
Invoke CameraActivity using intent builder -> CameraActivity opens up -> Click on take picture button -> takes picture and previews it for a second -> Rather than returning the picture to calling code, live camera preview resumes with the take picture button pressed -> take picture button now does not do anything, however, clicking android-back button works -> calling app is now frozen and android's 'app not responding' dialog shows up.

I'm using the latest version 0.7.2. However, I just noticed that an older version (0.6.4) which I was using in a previous app worked fine and didn't cause the freezing issue. I didn't check with any other versions between 0.6.4 and 0.7.2.

This is the code I'm using - 

```
Uri outputFileUri;
final File root = new File(Environment.getExternalStorageDirectory() + File.separator + "testApp" + File.separator);
root.mkdirs();
String timeStamp = new SimpleDateFormat("yyyyMMdd_Hms").format(new Date());
String fname = "JPEG_" + timeStamp + ".jpg";
final File sdImageMainDirectory = new File(root, fname);
outputFileUri = Uri.fromFile(sdImageMainDirectory);

CameraActivity.IntentBuilder cameraIntentBuilder = new CameraActivity.IntentBuilder(context)
        .skipConfirm()
        .to(sdImageMainDirectory)
        .debug()
        .facing(Facing.BACK)
        .updateMediaStore();

 Intent intent = cameraIntentBuilder.build();
 startActivityForResult(intent, CAMERA_INTENT_REQUEST_CODE);

//-------

public void onActivityResult(int requestCode, int resultCode, Intent data) {
    if (resultCode == Activity.RESULT_OK && requestCode == CAMERA_INTENT_REQUEST_CODE){
        if (outputFileUri != null) {
                    //Upload image to server
        }
}
```

Additional info - I'm using a OnePlus X 16GB. 

Thanks for this awesome library. You guys have really saved me tons of hours dealing with the edge cases of the camera API!

commonsguy, 2017-01-15:
I cannot reproduce the problem with the `demo-playground` app, using a OnePlus X, with equivalent settings to what your code snippet has.

Please try the `demo-playground` app and see if you can reproduce the problem. Also, if you go into Settings > About, what is the "Model number" value? Mine is "ONE E1005".

abhiank209, 2017-01-29:
Hello, 
Sorry for the delayed response. 

I cloned your project and built the demo-playground app and the issue is still happening on my device. The model number is "ONE E1003". However, the issue is not happening consistently, so I couldn't pinpoint a particular flow in which it was happening. I've attached the logcat output from the demo-playground app when the issue was happening. Hope this helps. 

[Logcat output from Cam2 library.txt](https://github.com/commonsguy/cwac-cam2/files/737331/Logcat.output.from.Cam2.library.txt)

commonsguy, 2017-01-29:
If, at some point, I can get my hands on an E1003, I will see if I can reproduce the problem. There is nothing in your LogCat that would give me anything to go on.

abhiank209, 2017-02-01:
Oh okay. Also, LeakCanary was reporting that MainActivity has leaked when the problem was happening.