commonsguy/cwac-cam2 #305: Unhandled NullPointerException setting display orientation

adambkaplan, 2017-02-02:
Device: Nexus 5X
Android Version: 7.1.1
Security Patch Level: January 5, 2017

Occurred after I invoked the video activity as follows:

```
VideoRecorderActivity.IntentBuilder videoBuilder = new VideoRecorderActivity.IntentBuilder(getContext())
    .durationLimit(duration)
    .to(videoFile);
startActivityForResult(videoBuilder.build(), REQUEST_CODE_VIDEO);
```

Forcing the orientation mode didn't seem to help.

Stack trace:

```
Could not dispatch event: class com.commonsware.cwac.cam2.CameraEngine$OrientationChangedEvent to subscribing class class com.commonsware.cwac.cam2.CameraController
    java.lang.RuntimeException: set display orientation failed
        at android.hardware.Camera.setDisplayOrientation(Native Method)
        at com.commonsware.cwac.cam2.plugin.OrientationPlugin$Classic.configureStillCamera(OrientationPlugin.java:149)
        at com.commonsware.cwac.cam2.ClassicCameraEngine$Session.configureStillCamera(ClassicCameraEngine.java:618)
        at com.commonsware.cwac.cam2.ClassicCameraEngine.handleOrientationChange(ClassicCameraEngine.java:409)
        at com.commonsware.cwac.cam2.CameraController.onEventMainThread(CameraController.java:427)
        at java.lang.reflect.Method.invoke(Native Method)
        at org.greenrobot.eventbus.EventBus.invokeSubscriber(EventBus.java:485)
        at org.greenrobot.eventbus.EventBus.postToSubscription(EventBus.java:420)
        at org.greenrobot.eventbus.EventBus.postSingleEventForEventType(EventBus.java:397)
        at org.greenrobot.eventbus.EventBus.postSingleEvent(EventBus.java:370)
        at org.greenrobot.eventbus.EventBus.post(EventBus.java:251)
        at com.commonsware.cwac.cam2.plugin.OrientationPlugin$1.onOrientationChanged(OrientationPlugin.java:55)
        at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
        at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:699)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:323)
        at android.os.Looper.loop(Looper.java:136)
        at android.app.ActivityThread.main(ActivityThread.java:6119)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
```

commonsguy, 2017-02-02:
I am unable to reproduce this using the `demo-playground/` app, with equivalent settings (using 5 seconds for the duration), on a Nexus 5X running 7.1.1.

- Are you using 0.7.3 of the library?
- Are you doing anything particular when you have the `VideoRecorderActivity` running?
- Are you able to reproduce this using `demo-playground/`?

adambkaplan, 2017-02-02:
Figured out the issue. I was setting the `.duration` to an extremely low value (30ms). The reaction time for the camera exceeds that.

closed 2017-02-02 by adambkaplan

commonsguy, 2017-02-02:
Whew. You had me scared for a bit there. :-) Thanks for the update!