commonsguy/cwac-cam2 #308: Toggle camera preview brightness

idish, 2017-02-19:
I have noticed the camera preview is dark compared to the stock camera, or other camera apps.
It would be great as if I could set the brightness level of the camera preview.

I took screenshots of the same item with the exact same surrounding light and brightness level of my device to show for the major differences between the previews:

Cwac-cam2 preview (yes I changed the preview buttons): 
[image]

Stock camera preview:
[image]

WaresTrader, 2017-02-19:
Oh

commonsguy, 2017-02-19:
I need the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem (see [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports)).

idish, 2017-02-19:
My device is Android 6.0 and the model number is 
> LG-H815L

Is it sufficient data?

commonsguy, 2017-02-19:
It is sufficient for me to identify the device. It is fairly unlikely that I will be able to obtain one, but I will keep an eye out for it.

commonsguy, 2017-02-19:
Also, please note that you do not appear to be using my library, as the screenshot does not reflect my library's UI. If you have modified my library, please test an unmodified library (e.g., test the standard `demo-playground/` app).

idish, 2017-02-20:
cwac-cam2 demo:
[image]

stock camera:
[image]

I will remove the original screenshots in the post as they are irrelevant anymore.

idish, 2017-04-01:
Well, 
I figured out it's not an hardware issue, but an API issue.

Here's the experiment I've made:
1. demo app with classic engine camera (default engine for my LG G4):
[image]
2. demo app with forced Camera2 engine (on low quality, because high quality doesn't work for me):
[image]
3. My stock camera
[image]

Of course all of these pictures were taken with the exact surrounding light in the room. 
Seems that the problems lays in the "classic" camera API..? Ahhh

idish, 2017-06-10:
Can the pending-hardware tag be removed, as it seems that it's not a hardware specific problem?

commonsguy, 2017-06-12:
By "pending-hardware", I mean "I need to get my hands on specific hardware to attempt to reproduce the problem". The good news is that I have that hardware now (I think). The bad news is that I need to cobble together the time to work on this.

commonsguy, 2017-06-18:
Actually, I have the LG-K350, which is the closest that I can get to the LG-H815L. And I get the opposite of what you reported: the classic API works and `camera2` is dark. The classic API will become the default again with v0.8.0 of the library, which will address the LG-K350 scenario. And the library, since v0.7.3, set up LG's default to be the classic API, so the LG-K350 is OK for now.

Re-test your LG-H815L and re-confirm your scenario: classic API is bad and `camera2` is good. In that case, I need you to provide me with the `Build.PRODUCT` value for your LG-H815L.

idish, 2017-06-28:
Sorry about the delay, been busy in the past days. 
Aww that's weird :O 
So I can confirm by 100% that `classic `API is bad, and `camera2 `is good regarding brightness in a very noticeable way unfortunately. (I've put leading breakpoints to each of the "force engine" options to make sure the required camera are indeed loaded)

So my device's `Build.PRODUCT` output is: p1_global_com.

I also checked some relevant `stackoverflow` posts regarding low brightness on dark. 
Maybe this would help in any way:
https://stackoverflow.com/questions/26967490/android-camera-preview-is-dark
https://stackoverflow.com/questions/31330098/android-surfaceview-preview-too-dark

Anyhow, it doesn't seem as that there's a general working solution :(

commonsguy, 2017-06-28:
OK, I'll steer this device to camera2 going forward. Thanks for the update!

idish, 2017-06-28:
Great, cheers!

idish, 2017-06-28:
Also - Not sure you have this in your mind, even though I mentioned it earlier: 
As of now, `camera2 `engine cannot be loaded when high quality is needed in this device. it can load only on low quality images (Checked it in the demo app). 
I remember debugging this and I saw that it cannot find a proper preview size for the requested image quality, so it just crashes without any warnings or ANRs (pretty heavy debugging). 
So it's not just a matter of constraints that will load the camera two engine by default for this device.

commonsguy, 2017-06-28:
Um, OK, then I'm back to not doing anything. I will continue to try to get my hands on this device, though I am not hopeful.