commonsguy/cwac-cam2 #309: [Question] Current limitations for implementing dynamic switching between picture/video?

lukaspili, 2017-02-28:
Hello,
Thanks for making that library and for the clean code base :)

I started a fork in order to implement dynamic switching between picture/video, without having to restart an activity. What are the current limitations for implementing such feature?

`CameraFragment` does not support it currently, but it seems to be more by design rather than technical limitation. 

The only part where it seems to be an issue is in the `FocusModePlugin` because: 

- `CameraSession` does not allow an update on its plugin chain;
- Continuous focus mode flag is different for picture and video

Is that correct? Do I miss something else?
Thanks for the help!

PS: I know and accept the consequences of using "unsupported APIs".

commonsguy, 2017-02-28:
This is an issue tracker, not a support board. I offer [multiple support options](https://github.com/commonsguy/cwac-cam2#questions). Please use them.

> Thanks for making that library and for the clean code base :)

This may be the second-worst pile of code that I have ever written for Android. The worst was the original CWAC-Camera library.

> What are the current limitations for implementing such feature?

I have never considered it, so I have no real idea, sorry.

> CameraFragment does not support it currently, but it seems to be more by design rather than technical limitation.

That seems like a fair assessment. Certainly there are camera apps that seem to offer toggling between the two without an activity change.

> `CameraSession` does not allow an update on its plugin chain

Correct, as there is no need for that in the current architecture.

> Do I miss something else?

I will be stunned if that is the only limitation in my code base for your desired functionality. But, as I noted, I have not given this a moment's thought, beyond the scope of this reply.

closed 2017-02-28 by commonsguy

lukaspili, 2017-02-28:
Sorry for posting my question at the wrong place.

I have assessed several camera apps and libraries, and I can assure you it's much more plaisant to dive into cwac-camera2 than any of the others.

Anyway, I will continue to play with cam2 and use stack-overflow for future communication if needed. Thanks for the quick response!