commonsguy/cwac-cam2 #310: Camera preview just won't open at all

shahzadyounas, 2017-03-03:
Hi

I have tried the latest build of cam2...and it no longer works on below:

Samsung BUILD.PRODUCT: serranoltexx (Samsung S4 Mini CyangenMod)

I downgraded in gradle to use cam2 0,6.6 and 0.6.9 and 0.7.0, and 0.7.1 and it worked fine.

But with latest cam2 0.7.5 and all the way down to 0.7.2, it doesn't:

I get these error messages:

```03-03 16:55:04.625 492-1657/? E/DatabaseUtils: Writing exception to parcel
                                               java.lang.SecurityException: Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL
                                                   at com.android.server.am.ActivityManagerService.handleIncomingUser(ActivityManagerService.java:15174)
                                                   at android.app.ActivityManager.handleIncomingUser(ActivityManager.java:2492)
                                                   at com.android.providers.settings.SettingsProvider.call(SettingsProvider.java:688)
                                                   at android.content.ContentProvider$Transport.call(ContentProvider.java:325)
                                                   at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:275)
                                                   at android.os.Binder.execTransact(Binder.java:404)
                                                   at dalvik.system.NativeStart.run(Native Method)
03-03 16:55:04.705 184-184/? E/SMD: DCD ON
03-03 16:55:04.895 191-804/? E/qdmemalloc: heap_msk=3000000 flags=1
                                            
03-03 16:55:04.905 191-804/? E/qdmemalloc: heap_msk=40000000 flags=1
                                            
03-03 16:55:05.096 13553-13661/com.muzmatch.muzmatchapp E/com.newrelic.agent.android: Total trace exclusive time is zero. Ignoring trace Display ConfirmationFragment
03-03 16:55:05.526 492-1090/? E/Watchdog: !@Sync 84
```

Or full:

```
03-03 16:55:33.115 492-844/? I/InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.135 ] when=2572456373000
03-03 16:55:33.115 492-843/? I/InputDispatcher: Delivering touch to: action: 0x4, toolType: 1
03-03 16:55:33.125 492-843/? I/InputDispatcher: Delivering touch to: action: 0x0, toolType: 1
03-03 16:55:33.125 492-812/? I/ServiceKeeper: In getpackagename pid = 492 uid = 1000 package name = android
                                              
                                              --------- beginning of /dev/log/system
03-03 16:55:33.125 492-843/? D/PowerManagerService: [api] userActivityFromNative : 10 (event: 2 flags: 0) eventTime = 2572456
03-03 16:55:33.165 492-844/? I/InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=2572505328000
03-03 16:55:33.165 492-843/? I/InputDispatcher: Delivering touch to: action: 0x1, toolType: 1
03-03 16:55:33.185 492-506/? E/DatabaseUtils: Writing exception to parcel
                                              java.lang.SecurityException: Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL
                                                  at com.android.server.am.ActivityManagerService.handleIncomingUser(ActivityManagerService.java:15174)
                                                  at android.app.ActivityManager.handleIncomingUser(ActivityManager.java:2492)
                                                  at com.android.providers.settings.SettingsProvider.call(SettingsProvider.java:688)
                                                  at android.content.ContentProvider$Transport.call(ContentProvider.java:325)
                                                  at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:275)
                                                  at android.os.Binder.execTransact(Binder.java:404)
                                                  at dalvik.system.NativeStart.run(Native Method)
03-03 16:55:33.185 492-506/? W/ActivityManager: Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL
03-03 16:55:33.206 492-12824/? D/CustomFrequencyManagerService: acquireDVFSLockLocked : type : DVFS_MIN_LIMIT  frequency : 1242000  uid : 1000  pid : 492  pkgName : ACTIVITY_RESUME_BOOSTER@5
03-03 16:55:33.206 492-12824/? W/ActivityManager: mDVFSHelper.acquire()
03-03 16:55:33.216 492-699/? D/PointerIcon: setMouseIconStyle1 pointerType: 1001iconType:101 flag:0
03-03 16:55:33.216 492-699/? D/PointerIcon: setMouseCustomIcon IconType is same.101
03-03 16:55:33.226 492-986/? V/WindowOrientationListener: ScreenOrientationEventListenerImpl.getProposedRotationLocked, Rotation: -1
03-03 16:55:33.226 492-986/? V/WindowOrientationListener: ScreenOrientationEventListenerImpl.getProposedRotationLocked, Rotation: -1
03-03 16:55:33.226 492-986/? V/WindowManager: rotationForOrientationLw(orient=-1, last=0); user=0  sensorRotation=-1 mLidState=-1 mDockMode=0 mHdmiPlugged=false mAccelerometerDefault=false gripRotationLock=false
03-03 16:55:33.246 492-812/? I/ServiceKeeper: In getpackagename pid = 492 uid = 1000 package name = android
03-03 16:55:33.246 2872-2872/? I/SecureStorage: [INFO]: SPID(0x00000000)Checking if this device supports Secure Storage
03-03 16:55:33.246 2872-2872/? I/SecureStorage: [INFO]: SPID(0x00000000)This device does not support Secure Storage (length of property)
03-03 16:55:33.266 492-1657/? V/WindowOrientationListener: ScreenOrientationEventListenerImpl.getProposedRotationLocked, Rotation: -1
03-03 16:55:33.266 492-1657/? V/WindowOrientationListener: ScreenOrientationEventListenerImpl.getProposedRotationLocked, Rotation: -1
03-03 16:55:33.266 492-1657/? V/WindowManager: rotationForOrientationLw(orient=7, last=0); user=0  sensorRotation=-1 mLidState=-1 mDockMode=0 mHdmiPlugged=false mAccelerometerDefault=false gripRotationLock=false
03-03 16:55:33.276 941-941/? D/PhoneStatusBar: setTransGradationMode=false, mTransparentMode=false, mSemiTransparentMode=false, mMultiWindowMode=false
03-03 16:55:33.276 492-506/? D/StatusBarManagerService: tr p:13553,o:f
03-03 16:55:33.356 13553-13553/com.muzmatch.muzmatchapp D/ProgressBar: setProgressDrawable drawableHeight = 18
03-03 16:55:33.356 13553-13553/com.muzmatch.muzmatchapp D/AbsSeekBar: AbsSeekBar Constructor: misSeebarAnimationAvailable = false
03-03 16:55:33.386 941-941/? D/PhoneStatusBar: setSemiTransparentMode=false, mTransparentMode=false, mSemiTransparentMode=false, mMultiWindowMode=false
[... 14 lines cut ...]
03-03 16:55:33.456 191-313/? E/qdmemalloc: heap_msk=40000000 flags=1
                                            
03-03 16:55:33.466 492-986/? V/WindowOrientationListener: ScreenOrientationEventListenerImpl.getProposedRotationLocked, Rotation: -1
03-03 16:55:33.466 492-986/? V/WindowOrientationListener: ScreenOrientationEventListenerImpl.getProposedRotationLocked, Rotation: -1
03-03 16:55:33.466 492-986/? V/WindowManager: rotationForOrientationLw(orient=1, last=0); user=0  sensorRotation=-1 mLidState=-1 mDockMode=0 mHdmiPlugged=false mAccelerometerDefault=false gripRotationLock=false
03-03 16:55:33.476 492-800/? D/STATUSBAR-StatusBarManagerService: manageDisableList what=0x0 pkg=WindowManager.LayoutParams
03-03 16:55:33.476 941-941/? D/PhoneStatusBar: setSemiTransparentMode=false, mTransparentMode=false, mSemiTransparentMode=false, mMultiWindowMode=false
03-03 16:55:33.476 492-1154/? D/StatusBarManagerService: semi p:13553,o:f
03-03 16:55:33.486 492-719/? D/CustomFrequencyManagerService: acquireDVFSLockLocked : type : DVFS_MIN_LIMIT  frequency : 1242000  uid : 1000  pid : 492  pkgName : ACTIVITY_RESUME_BOOSTER@9
03-03 16:55:33.486 492-699/? D/CustomFrequencyManagerService: releaseDVFSLockLocked : Getting Lock type frm List : DVFS_MIN_LIMIT  frequency : 1242000  uid : 1000  pid : 492  tag : ACTIVITY_RESUME_BOOSTER@5
03-03 16:55:33.486 492-699/? W/ActivityManager: mDVFSHelper.release()
03-03 16:55:33.496 492-699/? D/PointerIcon: setMouseIconStyle1 pointerType: 1001iconType:101 flag:0
03-03 16:55:33.496 492-699/? D/PointerIcon: setMouseCustomIcon IconType is same.101
03-03 16:55:33.496 191-804/? I/SurfaceFlinger: id=169 Removed DameraActiv (13/14)
03-03 16:55:33.496 191-321/? I/SurfaceFlinger: id=169 Removed DameraActiv (-2/14)
[... cut ...]
03-03 16:55:33.786 492-492/? D/CustomFrequencyManagerService: releaseDVFSLockLocked : Getting Lock type frm List : DVFS_MIN_LIMIT  frequency : 1242000  uid : 1000  pid : 492  tag : ACTIVITY_RESUME_BOOSTER@9
03-03 16:55:34.096 13553-13661/com.muzmatch.muzmatchapp E/com.newrelic.agent.android: Total trace exclusive time is zero. Ignoring trace Display ConfirmationFragment
03-03 16:55:34.157 941-941/? D/STATUSBAR-NetworkController: onSignalStrengthsChanged signalStrength=SignalStrength: 20 -1 -1 -1 -1 -1 -1 99 2147483647 2147483647 2147483647 -1 2147483647 gsm|lte 0x4 level=4
```

Appreciate your help on this! Tx (FYI a while back you fixed an issue for me for this phone model)

commonsguy, 2017-03-03:
Nothing in any of that LogCat output seems related to my library &mdash; are there specific messages in there that are tied to your app's process that I should be looking at?

I also do not know what "just won't open at all means".

Have you tried either of the demo apps?

> FYI a while back you fixed an issue for me for this phone model

I think I have that device, but I do not have CyanogenMod on it.

shahzadyounas, 2017-03-03:
Yes - Literally i tap to open the preview and nothing at all happens. That is the only stuff in the logs.

Downgrading cam2 to 0.7.1 and it works a treat. I tried every single release from current down to that, and all failed as above.

shahzadyounas, 2017-03-03:
I bet if you try the latest release on the device you have it will fail.

shahzadyounas, 2017-03-03:
FYI exact same results when using demo apps - v0.7.1 and below it works only.

commonsguy, 2017-03-03:
I blocked the front-facing camera on that device, as I was getting invalid output. See [this commit](https://github.com/commonsguy/cwac-cam2/commit/b51b539eaf5d5fcda36fee63636e0fdfd372c397). The experience that you will get is what you describe: the `CameraActivity` will `finish()` immediately.

For the rear-facing camera, both the `demo/` and the `demo-playground/` app work fine for me on that hardware, running Samsung's edition of Android 4.4.2 (not CyanogenMod/LineageOS).

See issue #311 for providing a cleaner way for you to find out that some camera/device combination is not supported.

closed 2017-03-03 by commonsguy

shahzadyounas, 2017-03-03:
Hi

But why would the same FFC work by default with 0.7.1 fine? IE with that release it gives good output?

```  int numCameras = Camera.getNumberOfCameras();

        if (numCameras > 1) {
            //we must have a FFC
            Intent i = new CameraActivity.IntentBuilder(this)
                    .facing(Facing.FRONT)
                    .facingExactMatch()
                    .to(new File(getCacheDir(), "selfie-portrait.jpg"))
                    .skipConfirm()
                    .orientationLockMode(OrientationLockMode.PORTRAIT)
                    .zoomStyle(ZoomStyle.NONE)
                    .updateMediaStore()
                    .build();

            startActivityForResult(i, REQUEST_PORTRAIT_FFC);

        } else {
            Intent i = new CameraActivity.IntentBuilder(this)
                    .facing(Facing.BACK)
                    .facingExactMatch()
                    .to(new File(getCacheDir(), "selfie-portrait.jpg"))
                    .skipConfirm()
                    .orientationLockMode(OrientationLockMode.PORTRAIT)
                    .zoomStyle(ZoomStyle.NONE)
                    .updateMediaStore()
                    .build();

            startActivityForResult(i, REQUEST_PORTRAIT_FFC);
        }```

This exact code with release 0.7.1 works fine and loads FFC correctly no issues. Any release above that doesnt at all.

commonsguy, 2017-03-03:
> But why would the same FFC work by default with 0.7.1 fine? 

It didn't on my test hardware.

shahzadyounas, 2017-03-03:
(purely out of interest if you tried 0.6.8 does it work for you?

Sorry to be a pain - just trying to understand as something strange is definitely happening.

shahzadyounas, 2017-03-03:
FYI on latest release yes rear camera loads fine (just tried)

commonsguy, 2017-03-03:
See issue #312 for retesting `serranoltexx`. I will not get to this for a week or two, at best, though.

shahzadyounas, 2017-03-03:
No worries - your work has been invaluable to us - so it is much appreciated. Thx for the speedy responses.

shahzadyounas, 2017-03-12:
Hi - just wondering if you had a chance to look at this one? Appreciate you have alot on your plate!