commonsguy/cwac-cam2 #312: Retest serranoltexx

commonsguy, 2017-03-03:
Issue #310 disputes the changes that I made to block FFC on `serranoltexx`. Retest that scenario, confirm whether or not FFC works reliably.

shahzadyounas, 2017-03-28:
Hi

Sorry to be a pain -  any luck on this? Samsung s4 is still (sadly) a popular phone!

Thanks v much in advance

commonsguy, 2017-03-28:
I have been busy dealing with Android Studio 2.3 and the O Developer Preview for my book, plus delivering training. As such, I have not had time to devote to this issue, nor will I for at least a couple of more weeks.

shahzadyounas, 2017-03-28:
Totally understand - appreciate all you can do thank you

commonsguy, 2017-04-09:
Well, the FFC is pretty bad, particularly in portrait mode, for previews, pictures, and videos. Hence, I am leaving this restriction in place.

closed 2017-04-09 by commonsguy

shahzadyounas, 2017-04-10:
Out of interest - can you let me know the code/place which causes the restriction? so I can fork it as I need it to work for that phone model.

Thanks!

commonsguy, 2017-04-10:
@shahzadyounas See [these lines](https://github.com/commonsguy/cwac-cam2/blob/v0.7.5/cam2/src/main/java/com/commonsware/cwac/cam2/CameraConstraints.java#L163-L169).

shahzadyounas, 2017-04-10:
Thanks - also last quick q  -method supportsCameraTwo - what is that for?

commonsguy, 2017-04-10:
As the name suggests, it indicates whether to use the `camera2` APIs, if the device is eligible for them.

If you have further questions, please use [the support options that I offer](https://github.com/commonsguy/cwac-cam2#questions).

shahzadyounas, 2017-04-10:
Gotcha - seems obvious looking at that now! Thx for your help