commonsguy/cwac-cam2 #314: NPE on Samsung Galaxy S7 running Android 7.0

op-and, 2017-03-10:
I am not forcing the CameraEngine to classic. I did see you made changes in code to set CameraEngine to Classic for LG phones - was that only for the phones specified in the constraints or all LG phones (Off topic, but if it is meant for all LG phones, since the constraints will return null for any LG phone not in the list, it will use Camera2). 

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'com.commonsware.cwac.cam2.CameraDescriptor com.commonsware.cwac.cam2.CameraSession.getDescriptor()' on a null object reference
       at com.commonsware.cwac.cam2.CameraTwoEngine.zoomTo(CameraTwoEngine.java:392)
       at com.commonsware.cwac.cam2.CameraController.handleZoom(CameraController.java:286)
       at com.commonsware.cwac.cam2.CameraController.setZoom(CameraController.java:275)
       at com.commonsware.cwac.cam2.CameraFragment$7.onProgressChanged(CameraFragment.java:695)
       at android.widget.SeekBar.onProgressRefresh(SeekBar.java:93)
       at android.widget.ProgressBar.doRefreshProgress(ProgressBar.java:1432)
       at android.widget.ProgressBar.refreshProgress(ProgressBar.java:1491)
       at android.widget.ProgressBar.setProgressInternal(ProgressBar.java:1556)
       at android.widget.AbsSeekBar.setProgressInternal(AbsSeekBar.java:1517)
       at android.widget.AbsSeekBar.trackTouchEvent(AbsSeekBar.java:1410)
       at android.widget.AbsSeekBar.startDrag(AbsSeekBar.java:1353)
       at android.widget.AbsSeekBar.onTouchEvent(AbsSeekBar.java:1293)
       at android.view.View.dispatchTouchEvent(View.java:10725)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2492)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2492)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2492)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2492)
       at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:505)
       at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1863)
       at android.app.Activity.dispatchTouchEvent(Activity.java:3226)
       at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:467)
       at android.view.View.dispatchPointerEvent(View.java:10954)
       at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:5051)
       at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4908)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4439)
       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4492)
       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4458)
       at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4591)
       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4466)
       at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4648)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4439)
       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4492)
       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4458)
       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4466)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4439)
       at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6936)
       at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6875)
       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6836)
       at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:7046)
       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:323)
       at android.os.Looper.loop(Looper.java:136)
       at android.app.ActivityThread.main(ActivityThread.java:6688)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1468)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1358)
```

commonsguy, 2017-03-10:
I need the `Build.MANUFACTURER` and `Build.PRODUCT` values for your device. I also need to know how to reproduce this problem. All of this is covered in [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#how-to-contribute).

op-and, 2017-03-13:
Crash was reported via Crashlytics. I will update with details once I can find a Samsung S7 and reproduce the issue.

commonsguy, 2017-03-13:
Crashlytics doesn't give you `Build` values? Even ACRA does that, and it's open source...

op-and, 2017-03-13:
Unfortunately, this is all Crashlytics will tell me:

# OS Version: 7.0
# Device: SAMSUNG-SM-G930A
# RAM Free: 24.2%
# Disk Free: 34.2%

commonsguy, 2017-03-13:
Wow, Crashlytics is weaker than I thought. Anyway, that's enough for me to isolate the specific model.

My guess, by eyeballing the exception, is that the activity was destroyed and recreated (e.g., user left and came back ~10 minutes later, after the process had been terminated), and I'm not handling that restoration properly. I'll look into this. If you happen to come upon a reproducible scenario, though, please let me know.

op-and, 2017-03-13:
Thank you. If I will update the issue if I get my hands on a phone and recreate the issue. 

BTW, my side-question re. forcing LGE phones to use camera 1 - was that intended for phones specified in the constraints or for all LGE phones?

commonsguy, 2017-03-13:
At the moment, that is intended for the phones listed in the constraints.

commonsguy, 2017-05-01:
Addressed in v0.7.6.

closed 2017-05-01 by commonsguy