commonsguy/cwac-cam2 #315: Breaking incompatibility with com.android.support:design:25.2.0

rusxakep, 2017-03-13:
java.lang.NoSuchFieldError: FloatingActionButton_backgroundTint
	at android.support.design.widget.FloatingActionButton.<init>(FloatingActionButton.java:164)
	at android.support.design.widget.FloatingActionButton.<init>(FloatingActionButton.java:153)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.jetbrains.android.uipreview.ViewLoader.createNewInstance(ViewLoader.java:475)
	at org.jetbrains.android.uipreview.ViewLoader.loadClass(ViewLoader.java:262)
	at org.jetbrains.android.uipreview.ViewLoader.loadView(ViewLoader.java:220)
	at com.android.tools.idea.rendering.LayoutlibCallbackImpl.loadView(LayoutlibCallbackImpl.java:186)
	at android.view.BridgeInflater.loadCustomView(BridgeInflater.java:334)
	at android.view.BridgeInflater.loadCustomView(BridgeInflater.java:345)
	at android.view.BridgeInflater.createViewFromTag(BridgeInflater.java:245)
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:727)
	at android.view.LayoutInflater.rInflate_Original(LayoutInflater.java:858)
	at android.view.LayoutInflater_Delegate.rInflate(LayoutInflater_Delegate.java:70)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:834)
	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:821)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:518)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:397)
	at com.android.layoutlib.bridge.impl.RenderSessionImpl.inflate(RenderSessionImpl.java:324)
	at com.android.layoutlib.bridge.Bridge.createSession(Bridge.java:429)
	at com.android.ide.common.rendering.LayoutLibrary.createSession(LayoutLibrary.java:368)
	at com.android.tools.idea.rendering.RenderTask$2.compute(RenderTask.java:567)
	at com.android.tools.idea.rendering.RenderTask$2.compute(RenderTask.java:549)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:863)
	at com.android.tools.idea.rendering.RenderTask.createRenderSession(RenderTask.java:549)
	at com.android.tools.idea.rendering.RenderTask.lambda$inflate$1(RenderTask.java:680)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

Reproduce:

Download official demo of support library: https://github.com/chrisbanes/cheesesquare
Add to project depends: compile 'com.commonsware.cwac:cam2:0.7.5'

Build and try open layout include_list_viewpager.xml in android studio 2.3 gui editor.

rusxakep, 2017-03-13:
I think problem with: "The cam2 artifact depends upon **com.github.clans:fab (for a floating action button and floating action menu implementation)**

Please upgrade depends, if possible.

rusxakep, 2017-03-13:
I found workaround, need replace tag

android.support.design.widget.FloatingActionButton

To

com.github.clans.fab.FloatingActionButton

commonsguy, 2017-03-13:
> Breaking incompatibility with com.android.support:design:25.2.0

FWIW, the cited cheesesquare app does not use that version of that library.

> Build and try open layout include_list_viewpager.xml in android studio 2.3 gui editor.

I did not write Android Studio, let alone its layout editor. I did not write `android.support.design.widget.FloatingActionButton`. I have no means of addressing problems in either of those products.

Please follow [this issue](https://code.google.com/p/android/issues/detail?id=235977) to learn more about if/when Google addresses the bug in their IDE.

Note that the cheesesquare app runs just fine. The problem is limited to the layout editor, which only occasionally works.

Also note that you get the same problem by eliminating the `cam2` dependency and depending directly on the `clans` FAB library. IOW, this has nothing to do with me.

> Please upgrade depends, if possible.

This library already uses the latest edition of that dependency.

> I found workaround

That is not really a workaround.

closed 2017-03-13 by commonsguy