commonsguy/cwac-cam2 #319: ImageContext.getJpeg(true) on Samsung Galaxy S4 Mini results in either OOM or larger image post-re-orientation

duckinferno, 2017-03-27:
Bit of a mouthful of a title, sorry.

My Samsung Galaxy S4 Mini running 4.4.2 fails after 1-2 snaps with an OutOfMemoryException within the `ImageContext.getJpeg(bool)` method, sometimes during the `rotateViaMatrix` step, other times during the `writeExif` step.

```
E/CWAC-Cam2: Exception in camera processing
             java.lang.OutOfMemoryError
                 at android.graphics.Bitmap.nativeCreate(Native Method)
                 at android.graphics.Bitmap.createBitmap(Bitmap.java:928)
                 at android.graphics.Bitmap.createBitmap(Bitmap.java:901)
                 at android.graphics.Bitmap.createBitmap(Bitmap.java:833)
                 at com.commonsware.cwac.cam2.ImageContext.rotateViaMatrix(ImageContext.java:250)
                 at com.commonsware.cwac.cam2.ImageContext.getJpeg(ImageContext.java:105)
                 at com.commonsware.cwac.cam2.JPEGWriter.process(JPEGWriter.java:77)
                 at com.commonsware.cwac.cam2.PictureTransaction.process(PictureTransaction.java:38)
                 at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:525)
                 at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                 at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                 at java.lang.Thread.run(Thread.java:841)
```

I can guess at a few possible causes:
1. Up to three instances of the data can exist in memory at once (vars `jpegOriginal`, `original`, `rotated`) before the image is compressed
2. No options are passed in to `BitmapFactory.decodeByteArray()` to control its file size; older devices like the S4 Mini tend to have cameras that are 'too powerful' for apps to load into memory 1:1 without issues like this
3. The below issue likely also contributes to the excess of data in memory

In addition, on the occasions in which it succeeds, the size of the bitmap is roughly doubled after its orientation change, ie. jpegOriginal contains 2.2mb of data at the end of the method, and 4.4mb at the end.  It also takes a few too many moments to process, which feels "too long" as a user.

If 'false' is supplied to `getJpeg(bool)`, none of the above occurs - the picture is taken snappily and is of a reasonable size.

commonsguy, 2017-03-27:
This is an issue tracker, not a support board. Please use the support resources outlined in [the documentation](https://github.com/commonsguy/cwac-cam2#questions).

You can disable image orientation normalization by calling `skipOrientationNormalization()` on the `CameraActivity.IntentBuilder`. This should clear up your problem, at the cost of you now having to rely upon the EXIF orientation tag to know how to display the image.

closed 2017-03-27 by commonsguy

duckinferno, 2017-03-27:
The intent of this was to raise an issue with the above method, which causes an OOM on this device and doubles the size of the original image.  

We did end up using `skipOrientationNormalization()` as a work around.