commonsguy/cwac-cam2 #320: TextureView doesn't support displaying a background drawable

mraonghus, 2017-04-16:
Hi,

First off, thanks for being such a force in the Android community, your book and your Stack Overflow output is second to none.

I adapted my original code using the standard Android Camera API to cwac-cam2 and it works really well on the first phone I tried (Xiaomi Redmi 4, Marshmallow), but is crashing once I launch CameraActivity on the second phone (Xiaomi Mi5, Nougat). I've followed the tutorial in your book for the implementation mostly. It was painless save for finding the appropriate theme (i.e. with an Action Bar).

Here's the stack trace:

```
04-16 17:06:24.155 32219-32219/com.foodmarble.pressdemo E/AndroidRuntime: FATAL EXCEPTION: main
                                                                          Process: com.foodmarble.pressdemo, PID: 32219
                                                                          java.lang.RuntimeException: Unable to start activity ComponentInfo{com.foodmarble.pressdemo/com.commonsware.cwac.cam2.CameraActivity}: android.view.InflateException: Binary XML file line #12: Binary XML file line #12: Error inflating class com.commonsware.cwac.cam2.CameraView
                                                                              at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2688)
                                                                              at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2749)
                                                                              at android.app.ActivityThread.-wrap12(ActivityThread.java)
                                                                              at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1498)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                              at android.os.Looper.loop(Looper.java:160)
                                                                              at android.app.ActivityThread.main(ActivityThread.java:6200)
                                                                              at java.lang.reflect.Method.invoke(Native Method)
                                                                              at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:874)
                                                                              at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:764)
                                                                           Caused by: android.view.InflateException: Binary XML file line #12: Binary XML file line #12: Error inflating class com.commonsware.cwac.cam2.CameraView
                                                                           Caused by: android.view.InflateException: Binary XML file line #12: Error inflating class com.commonsware.cwac.cam2.CameraView
                                                                           Caused by: java.lang.reflect.InvocationTargetException
                                                                              at java.lang.reflect.Constructor.newInstance0(Native Method)
                                                                              at java.lang.reflect.Constructor.newInstance(Constructor.java:430)
                                                                              at android.view.LayoutInflater.createView(LayoutInflater.java:652)
                                                                              at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:812)
                                                                              at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:752)
                                                                              at android.view.LayoutInflater.rInflate(LayoutInflater.java:883)
                                                                              at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:846)
                                                                              at android.view.LayoutInflater.rInflate(LayoutInflater.java:886)
                                                                              at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:846)
                                                                              at android.view.LayoutInflater.inflate(LayoutInflater.java:522)
                                                                              at android.view.LayoutInflater.inflate(LayoutInflater.java:430)
                                                                              at com.commonsware.cwac.cam2.CameraFragment.onCreateView(CameraFragment.java:249)
                                                                              at android.app.Fragment.performCreateView(Fragment.java:2352)
                                                                              at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:995)
                                                                              at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1171)
                                                                              at android.app.BackStackRecord.run(BackStackRecord.java:815)
                                                                              at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1580)
                                                                              at android.app.FragmentController.execPendingActions(FragmentController.java:371)
                                                                              at android.app.Activity.performStart(Activity.java:6761)
                                                                              at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2651)
                                                                              at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2749)
                                                                              at android.app.ActivityThread.-wrap12(ActivityThread.java)
                                                                              at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1498)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                              at android.os.Looper.loop(Looper.java:160)
                                                                              at android.app.ActivityThread.main(ActivityThread.java:6200)
                                                                              at java.lang.reflect.Method.invoke(Native Method)
                                                                              at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:874)
                                                                              at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:764)
                                                                           Caused by: java.lang.UnsupportedOperationException: TextureView doesn't support displaying a background drawable
                                                                              at android.view.TextureView.setBackgroundDrawable(TextureView.java:302)
                                                                              at android.view.View.setBackground(View.java:18130)
                                                                              at android.view.View.<init>(View.java:4574)
                                                                              at android.view.View.<init>(View.java:4081)
                                                                              at android.view.TextureView.<init>(TextureView.java:159)
                                                                              at com.commonsware.cwac.cam2.CameraView.<init>(CameraView.java:66)
                                                                              at java.lang.reflect.Constructor.newInstance0(Native Method) 
[... cut ...]
                                                                              at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:874) 
                                                                              at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:764)
```

I suspect the relevant line is:
`Caused by: java.lang.UnsupportedOperationException: TextureView doesn't support displaying a background drawable`

Any help much appreciated!

Thanks,
Aonghus

commonsguy, 2017-04-16:
> First off, thanks for being such a force in the Android community, your book and your Stack Overflow output is second to none.

Um, thanks!

> I suspect the relevant line is: `Caused by: java.lang.UnsupportedOperationException: TextureView doesn't support displaying a background drawable`

Yes, but I do not use a background drawable. Did you modify the library?

> It was painless save for finding the appropriate theme (i.e. with an Action Bar).

If you replaced the theme used by library, try removing that and see if the error persists.

mraonghus, 2017-04-16:
Thanks for the super quick response.

I didn't modify the library, but by changing the theme I was able to fix the error. Thank you!

Previously I had this:
````
<style name="CameraThemeSelector" parent="Widget.AppCompat.ActionBar">
    <item name="android:background">@color/black_overlay</item>
</style>
```
After changing to this:

```
<style name="CameraThemeSelector" parent="Widget.AppCompat.ActionBar" />
```

It now works. However, I'm not sure now how to achieve a black background behind the photo after it is captured. Is there a way to do that?

Thanks again!
Aonghus

commonsguy, 2017-04-16:
> However, I'm not sure now how to achieve a black background behind the photo after it is captured. Is there a way to do that?

Do that in your own activity that is showing the photo after it is captured.

Don't apply that style to my activity. For starters, I am not using `appcompat-v7`, and so a `Widget.AppCompat` style is likely to cause other problems.

closed 2017-04-16 by commonsguy

mraonghus, 2017-04-16:
Sorry, I thought this was the right place for this sort of query. Noted.

My problem is having a dark background where the user choose to accept or re-take the photo. I should be able to figure it out. Thanks.

commonsguy, 2017-04-16:
In principle, you should be able to control the background by switching the theme, but base it off of a stock theme (e.g., `Theme.Material.Light`). See how I am overriding the themes in [the `demo-playground` app](https://github.com/commonsguy/cwac-cam2/blob/v0.7.5/demo-playground/src/main/AndroidManifest.xml).

mraonghus, 2017-04-16:
Thanks again for your help. I think I've found a workable solution 👍